MarlinFirmware/Marlin #20204: [BUG]probe wizard homes n goes back to main screen (short description)

cvanbusk, 2020-11-19:
Latest version of bugfix, and config. Was working yesterday when I complied firmware. I'll try again tonight, but hopefully someone has had this issue n can point me in the right direction. 

Please follow the instructions below. Failure to do so may result in your issue being closed.

### Before Reporting a Bug

1. Test with the `bugfix-2.0.x` 

2. Get troubleshooting help from the Marlin community to confirm it's a bug and not just a configuration error. Links at https://github.com/MarlinFirmware/Marlin/issues/new/choose

### Instructions

1. Fill out every section of the template below.

2. Always attach configuration files, regardless of whether you think they are involved.

3. Read and understand Marlin's Code of Conduct. By filing an Issue, you are expected to comply with it, including treating everyone with respect: https://github.com/MarlinFirmware/Marlin/blob/master/.github/code_of_conduct.md

-->

### Bug Description

### Configuration Files

**Required:** Include a ZIP file containing `Configuration.h` and `Configuration_adv.h`.

If you've made any other modifications describe them in detail here.

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:**

**Actual behavior:**

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [Contributing to Marlin](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md) for additional guidelines.

qwewer0, 2020-11-19:
@cvanbusk 

> * Fill out every section of the template below.
> * Always attach configuration files, regardless of whether you think they are involved.

qwewer0, 2020-11-19:
When I go to Probe Offset Wizard in the LCD

1. It shows "Homing XYZ"
2. Before homing is done the LCD goes back to the status screen
3. After it homed, it doesn't move to bed center (`PROBE_OFFSET_WIZARD_XY_POS`), stays at homing position
4. If I select Probe Offset Wizard again, LCD shows "Moving..."
5. It starts to move to home direction (X/Y min) until it hits one of the endstops
6. After it hit an endstop it starts to move to X/Y max and crashes into the frame (at this point I shot down the printer)

Ender 3, SKR Mini E3 v1.2, Latest Bugfix
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5558118/Configuration.zip)

https://github.com/MarlinFirmware/Marlin/pull/20167#issuecomment-729456456

cvanbusk, 2020-11-20:
1. either shows homing xyz 
2. Goes to Info Page showing ready if I go try again it show wizard setting for a short time then goes to info screen
Ender pro 3 Building with platforio latest n I tried 2.10 as that one i updated just before compiling. 

Thanks
RVB

[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/5570631/Configuration_adv.zip)

cvanbusk, 2020-11-20:
Loaded n configured 2.0.7 stable build works fine. Just n FYI.

qwewer0, 2020-11-20:
I think https://github.com/MarlinFirmware/Marlin/pull/20167 is the culprit, but I can't test right now.

@cvanbusk Could you test [Bugfix-2.0.x 50a77ef](https://github.com/MarlinFirmware/Marlin/tree/50a77ef7f05dd4ac8b5ed3e9f3e380c5da7ef50c)?

cvanbusk, 2020-11-25:
Trying it now, sorry for the delay

cvanbusk, 2020-11-25:
That works, Z-probe working as expected. Thank you

shitcreek, 2020-11-27:
Please close this issue if it's all dandy now.

rhapsodyv, 2020-11-30:
I'm closing this, as the users says it's fixed. Thanks.

closed 2020-11-30 by rhapsodyv