MarlinFirmware/Marlin #20210: [BUG] Confirming bed temperature on TFT_COLOR_UI sets it to 0

onfoot, 2020-11-19:
### Bug Description

When opening bed, nozzle temperature, or fan speed setting from the status screen on TFT_COLOR_UI running on TS35, tapping on the checkmark to confirm the setting sets it to 0, even if they are set to a non-zero value.

It works properly when accepting the setting using the rotary encoder.

Doesn't matter whether the value is changed using the rotary encoder, the slider, or the + button. Only the way of accepting the setting results in different behavior.

### Configuration Files

[Config.zip](https://github.com/MarlinFirmware/Marlin/files/5570064/Config.zip)

### Steps to Reproduce

1. Tap bed temperature setting from status screen
2. Change to 50
3. Tap checkmark to confirm

**Expected behavior:**

Bed temperature is set to 50

**Actual behavior:**

Bed temperature is set to 0

#### Additional Information

[Video](https://youtu.be/-zH1ZAWi95I)

rhapsodyv, 2020-11-20:
Are you using last bugfix?

I tested it and worked for me.

Section59, 2020-11-20:
Having this issue also using #define MKS_ROBIN_TFT35 - Setting it via octoprint seems to work, and shows on the screen as the correct set temp.

rhapsodyv, 2020-11-20:
Ok, found it. Fixed: #20213

onfoot, 2020-11-20:
Fix verified. 🥳

PR #20213 (2020-11-20): SINGLE_TOUCH_NAVIGATION follow up
### Description

Separe click actions to not break another controls.

### Benefits

Fix #20210
closed 2020-11-20 by rhapsodyv