MarlinFirmware/Marlin #20211: [BUG] g29 s1 doesn't keep home offset MBL bugfix-2.0.x 20000008

Sh8tan, 2020-11-19:
### Bug Description

When doing mesh bed leveling, g29 s1, does not start from home offset but from endstop. 

When doing g28 here is my positions are x=0 y=0 z=-350

### Configuration Files

Last bugfix-2.0.x 

### Steps to Reproduce

1. G28
2. Move z until +0.350
3. Set home offset
4. Store settings
5. G28
6. Mesh bed leveling
7. First probe point touch the bed until raise to 0.350

**Expected behavior:**

G29 s1 keep home offsets from Z to start

**Actual behavior:**

G29 s1 too low and doesn't keep z0 home offset but z0 endstop as reference
#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [Contributing to Marlin](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md) for additional guidelines.

boelle, 2020-11-20:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

Sh8tan, 2020-11-20:
My bad, here is my configs files, lines which were change are only in configuration.h

uncommented

#define PROBE_MANUALLY
#define MANUAL_PROBE_START_Z 0.2

#define RESTORE_LEVELING_AFTER_G28

and i change #define LEVEL_CORNERS_HEIGHT 0 ----> #define LEVEL_CORNERS_HEIGHT RAW_Z_POSITION (0.0)
[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/5573379/Configuration_adv.zip)
[Configuration.h.zip](https://github.com/MarlinFirmware/Marlin/files/5573380/Configuration.h.zip)

boelle, 2020-11-20:
good thing i added them as this is not the default

// Choose the name from boards.h that matches your setup
#ifndef MOTHERBOARD
  #define MOTHERBOARD BOARD_BTT_SKR_MINI_E3_V2_0
#endif

default is ramps 1.4 so you prob changed a lot more than those 2 lines

Vertabreak, 2020-11-20:
Set home offset https://marlinfw.org/docs/gcode/M206.html is part of the CNC system and without re-coding it to account for probe offsets it does not work with 3dprinters. as a work around if you account for probe offsets in addition to the number set by home offset manually with the gcode command you can get it to sorta work.

Sh8tan, 2020-11-20:
@boelle i already change my plateform in plateformio.ini to STM32F103RC_BTT_512K. Anwsering my own question to solve and close it :

Enable Z offset in lcd menu

1. Configuration.h

 #define PROBE_MANUALLY
 #define MANUAL_PROBE_START_Z 0.2

 #define FIX_MOUNTED_PROBE
 #define Z_SAFE_HOMING

2. Configuration_adv.h
 #define PROBE_OFFSET_WIZARD

3. SanityCheck.h
Commented line
 //#error "Z_SAFE_HOMING is recommended when homing with a probe. Enable it or comment out this line to continue."

then i choose the right offset from bed to nozzle and when doing a mesh bed level or level corners with the offset the right z0 a keeping. 

Thx for your help !

closed 2020-11-20 by Sh8tan