MarlinFirmware/Marlin #20215: [BUG] Calling G29 J while ZProbe/Endstop is triggered won't stop function

Protean-Man, 2020-11-20:
Having UBL active and calling G29 J to tilt the leveling mesh while ZProbe/Endstop is triggered will allow G29 J to run without min endstop limits being followed - which then causes bed crash and possible damage. 
This isn't a bug necessarily - just a missing stop call to ensure printer doesn't break itself. 

I found this out because updating to the new Bugfix had Z_AFTER_HOMING commented out by default for some reason. Which after homing the Z axis would keep the probe at bed level and hence - triggered. Uncommenting Z_AFTER_HOMING fixed the problem of my bed crashing through my carriage... but the issue of a safety stop call is still missing. 

I don't think you need my configs for this. It's pretty self explanatory. 

### Steps to Reproduce

1. commeng out Z_AFTER_HOMING so your Z doesn't lift after homing and stays active against the bed.
2. Have UBL active
3. Home Z and then make sure your Z endstop/probe is triggered with a M119 once it finishes homing. 
4. Call G29 J and watch your bed crash into your carriage.

**Expected behavior:**

If Z Probe/Endstop is triggered when any G29 call is made the printer will throw an error and stop.
-or-
If Z Probe/Endstop is triggered when any G29 call is made, the printer will lift the Z axis to fix itself before continuing with G29 X

**Actual behavior:**

Z crashes and doesn't stop until your carriage is destroyed. King Kong style destruction of your carriage.

#### Additional Information

Please ensure Z_AFTER_HOMING is not commented out by default to save others.

boelle, 2020-11-20:
The issue template says configs are required and i know that dev people will request them (no matter if you think they are requred or not) so:

Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

sjasonsmith, 2020-11-22:
@Protean-Man please post your configuration files. Normally I would just close this due to the missing files, but the rest of the report is well-written, and it sounds like an important issue to fix.

Providing your configuration files makes it easier for someone to reproduce and fix the issue.

Protean-Man, 2020-11-29:
No problem. 
[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/5612669/Configuration_adv.zip)

closed 2021-01-06 by github-actions[bot]