MarlinFirmware/Marlin #20220: [BUG] (MKS Robin nano_v2 - the touchscreen does not work)

Malderin, 2020-11-20:
I'm using the latest marlin 2.x bugfix. 
MKS robin nano V2 board.

### Bug Description

Everything worked two weeks ago. On the latest version of marlin, there is no reaction when pressing the touchscreen.

### Configuration Files

[configs.zip](https://github.com/MarlinFirmware/Marlin/files/5573251/configs.zip)

boelle, 2020-11-20:
can you find out precisely where it stopped working?

Malderin, 2020-11-20:
> can you find out precisely where it stopped working?

For 2 weeks there were a lot of changes, I tried to find, but failed

rhapsodyv, 2020-11-20:
We removed the touch calibration values from the board pins files. You need to copy it to your configs or use TOUCH_CALIBRATION_SCREEN to calibrate and save the values in the EEPROM.

Malderin, 2020-11-20:
> We removed the touch calibration values from the board pins files. You need to copy it to your configs or use TOUCH_CALIBRATION_SCREEN to calibrate and save the values in the EEPROM.

I am using mksui. I didn't really understand about TOUCH_CALIBRATION_SCREEN, so I did it like this:
Added to file configuration.h:
#if ENABLED(TFT_LVGL_UI)
  #define TOUCH_CALIBRATION_X -17253
  #define TOUCH_CALIBRATION_Y  11579
  #define TOUCH_OFFSET_X       514
  #define TOUCH_OFFSET_Y       -24
#endif
The touch screen is working.

rhapsodyv, 2020-11-20:
Exactly that! It was removed from pins file, so you need to keep it in your configs. 

`TOUCH_CALIBRATION_SCREEN` option will allow you to run M995, that will show up a screen to calibrate the touch. But if you have the values, you can use it.

Thanks!

closed 2020-11-20 by rhapsodyv