MarlinFirmware/Marlin #20221: [BUG] TMC2209 UART In dual Z mode E1 changes direction randomly

patlas, 2020-11-20:
### Bug Description

In dual Z mode for MKS Robin Nano V2 with TMC2209 controlled over UART second Z motor attached to E1 from time to time changes its direction when new move command send (no matter if over LCD or via PROTERFACE gcommand)

### Configuration Files
[ConfigHeaders.zip](https://github.com/MarlinFirmware/Marlin/files/5573281/ConfigHeaders.zip)

I have tested it in CoreXY mode and regular (separated motor for x and y)

### Steps to Reproduce

1. Start pronterface / LCD menu
2. Send Z move command (up or down) - multiple time

**Expected behavior:**

Both motor rotate the same direction or opposite (if inverted) but stable.

**Actual behavior:**

From time to time second Z motor changes it direction (randomly)

sjasonsmith, 2020-11-22:
Please test the [bugfix-2.0.x branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

If you still have issues please provide updated configuration files compatible with `bugfix-2.0.x`. Also include your pins file if you have modified it.

One troubleshooting step would be to switch the two driver modules to see if the problem stays with the socket or moves with the driver.

patlas, 2020-11-23:
Unfortunatelly I cannot build bugfix-2.0.x branch because each time I see: "#error "SERIAL_PORT is set to -1, but the MOTHERBOARD has no native USB support. Set SERIAL_PORT to a valid value for your board." I have tried both mks_robin_nano35 and mks_robin_nano35_stm32 with the same result. I attache my current config for bugfix-2.0.x branch.

[configs.zip](https://github.com/MarlinFirmware/Marlin/files/5585496/configs.zip)

sjasonsmith, 2020-11-23:
@patlas move the value from SERIAL_PORT_2 to SERIAL_PORT, then comment out SERIAL_PORT_2.

Those board don't have the USB Serial that -1 enables. We added the check recently because we realized many people enabled -1 because it is for USB, but have no good way to know whether their board actually can use it.

rhapsodyv, 2020-11-30:
@patlas did you tested bugfix?

rhapsodyv, 2020-12-02:
@patlas any news here? did you test bugfix?

sjasonsmith, 2020-12-24:
It has been a month with no response from submitter. Closing.

closed 2020-12-24 by sjasonsmith