MarlinFirmware/Marlin #20222: [BUG] Z Probe Past Bed with no probe defined

espr14, 2020-11-20:
### Bug Description

Is in `bugfix-2.0.x`, not in `2.0.x`.
No probe defined but in case of Z homing it states `Z Probe Past Bed`.

### Configuration Files

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5574366/Marlin.zip)

### Steps to Reproduce

1. undefine all probes
2. compile and load firmware
3. home Z axis

**Expected behavior:**

Should work

**Actual behavior:**

States `Z Probe Past Bed`

#### Additional Information

ellensp, 2020-11-21:
Cannot reproduce, Steps to reproduce unclear. Does not happen with provided config files.

1. this is clear.
2. I set the following, is this what you mean?
#define X_MIN_POS -10
#define Y_MIN_POS -10
3. this is clear
4. I set, the following, is this what you mean? 
  #define Z_SAFE_HOMING_X_POINT -2 // X point for Z homing
  #define Z_SAFE_HOMING_Y_POINT -2 // Y point for Z homing
5. It attempt to home Z, no errors displayed

please clarify the steps to reproduce.

ellensp, 2020-11-21:
Why do you need safe homing without a probe? It doesn't make sense. 
If you have safe homing the system thinks the Z endstop is mobile (ie attached to the extruder) in which case it is a probe of some sort.
Other wise the Z endstop is fixed to frame and Z should be able trigger it with X and Y in any reachable location.

espr14, 2020-11-21:
2. #define Z_SAFE_HOMING_X_POINT -5 // X point for Z homing
#define Z_SAFE_HOMING_Y_POINT -5 // Y point for Z homing
3.
4. G1 X-3 Y-3
M428
5. G28 Z

Z endstop can be at the bed level or even under it. This could depend on bed temperature. Save homing point ensures that nozzle does not touch or break the bed. Even with Z endstop above the bed you could get lower at the first touch due to fast homing speed.

rhapsodyv, 2020-11-22:
You have `#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN` enabled, so, for Marlin, you have a probe connected to Z Min. He interpret as you have a custom probe.

I can't see any reason to change any message. It's clear enough when using a probe, or when not using.

espr14, 2020-11-23:
Still the message is wrong. Let's say I have a probe connected to Z min and the probe is located at the right place (to the center of the bed). There's the same message still but the reason for the error is that the head cannot move behind the X/Y endstops (available area).

rhapsodyv, 2020-11-23:
The reason is that `Z_SAFE_HOMING` is supposed to home _inside_ the bed area. In the documentation:

// Use "Z Safe Homing" to **avoid homing with a Z probe outside the bed area.**

If you need a custom out of the bed homing, you have a very uncommon setup, and I don't think `Z_SAFE_HOMING` is for you. Maybe try using:
```
//#define MANUAL_X_HOME_POS 0
//#define MANUAL_Y_HOME_POS 0
//#define MANUAL_Z_HOME_POS 0
```

I will close this, as I don't think it's a bug. The message is really following the documentation and the intended use.

Anyway, feel free to send a PR if you think we need to change the message or the documentation.

Thanks for your help!

closed 2020-11-23 by rhapsodyv