MarlinFirmware/Marlin #20226: [PCF8574 i2c HD44780 2004] (RA_CONTROL_PANEL)

Cysign, 2020-11-20:
1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:**

**Actual behavior:**

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [Contributing to Marlin](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md) for additional guidelines.

ellensp, 2020-11-21:
This has already been fixed, but is only available in bugfix at this time.
See https://github.com/MarlinFirmware/Marlin/pull/19875

closed 2020-11-21 by ellensp

Cysign, 2020-11-21:
So I just have to download and use the bugfix-branch?

ellensp, 2020-11-21:
or apply the 2 small changes to your current code.
The differences https://github.com/MarlinFirmware/Marlin/pull/19875/files

Cysign, 2020-11-21:
Couldn't get it work... so I'm using a SSD1306 at the moment, which works properly ;)
Any idea how long it'll take until the bugfix is adopted to stable release?