MarlinFirmware/Marlin #20231: [BUG] Thermistors 998 & 999 make heated bed completely disappear from Marlin

Patag, 2020-11-21:
### Bug Description
Since few days, the heated bed disappeared from info screen and all related menus.
I'm using dummy thermistor 998 (tried also 999) as bed sensor on a test bench.

### Configuration Files
[config.zip](https://github.com/MarlinFirmware/Marlin/files/5577504/config.zip)

### Steps to Reproduce

1. #define TEMP_SENSOR_BED 998 (or 999) in configuration.h
2. Compile, download & run

**Expected behavior:**
Heated bed icon displayed on info screen with temp = 25° (or 100° if sensor 999).
All menus related to bed temperature displayed and functional.

**Actual behavior:**
None of the bed temperature related items are displayed

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [Contributing to Marlin](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md) for additional guidelines.

ellensp, 2020-11-21:
There was a PR that removed the need for dummy thermistors to need a pin, I suspect the lcd routines look for the existence of this pin as a flag to display the heated bed or not. https://github.com/MarlinFirmware/Marlin/pull/20159

ellensp, 2020-11-21:
yip confirmed
```
#if HAS_TEMP_BED && HAS_HEATER_BED
  #define HAS_HEATED_BED 1
``` 
is failing. 

To me it looks like HAS_ADC_TEST(P) in Marlin/src/inc/Conditionals_post.h isnt right... should be 
`#define HAS_ADC_TEST(P) ((PIN_EXISTS(TEMP_##P) || ENABLED(HEATER_##P##_DUMMY_THERMISTOR)) && TEMP_SENSOR_##P != 0 && DISABLED(HEATER_##P##_USES_MAX6675))
`
This compiles and the bed is back.
Needs further testing.

ellensp, 2020-11-21:
Fairly sure I'm correct, so created PR.

Patag, 2020-11-21:
Nice @ellensp !
May I close ?

ellensp, 2020-11-21:
please wait till pr gets merged, just in case there is an issue with it.

Patag, 2020-11-21:
No worries

sjasonsmith, 2020-11-22:
Thanks @Patag & @ellensp for reporting and investigating this.

When I made my initial change I had assumed that all the temp sensors were handled in the same way, and must have only tested with two extruders using dummy thermistors.

I'm working on fixing those inconsistencies, which will let my version of that `HAS_ADC_TEST` macro work for all thermistors. My intent was for that macro to mean "This actually needs a hardware ADC", which shouldn't be required for a dummy thermistor with no pin assignment.

sjasonsmith, 2020-11-23:
@Patag a solution has now been merged. I'll close this issue, but please reply if you encounter further issues with this.

closed 2020-11-23 by sjasonsmith

Patag, 2020-11-23:
All is running fine here.
Thanks @sjasonsmith !