MarlinFirmware/Marlin #20235: [BUG] Homing (safe) not working - BLTouch

C0unt23r0, 2020-11-21:
### Bug Description

Hotend chrashes into the bed or tries to exceed its limits!

Ender 3v2 - Controler 4.2.2
BL-Touch 3.0

Used latest bugfix-2.0.x (Where do i find the exact version? it was from 2020-11-19)  including the Ender3v2 configuration.

### Configuration Files

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5577675/Configuration.zip)

### Steps to Reproduce

1. Activate auto homing Or Auto Bed Leveling (ABL starts safe homing)
2. Disable Stepper
3. Move the hotend for example for manual bed leveling
4. Activate Auto Bed Leveling (maybe also start printing... not testest)

**Expected behavior:**

After disabling the stepper driver and activating the ABL again, i would expect that it starts the safe homing procedure again, but it doesnt't. As it is thinking that he knows its last position...

**Actual behavior:**

It starts the ABL right of, depending were i've left the hotend crashing into anything that is in its way. Even when it hits an endstop! Weather it is the X, or the Y Axes.

#### Additional Information

sjasonsmith, 2020-11-22:
I'm not really sure if ABL is significant for this. It might have issues with just simple homing and movement.

C0unt23r0, 2020-11-23:
You are right. I think it's about the logik behind the function homing_needed_error() including axes_should_home(). There are some slightly changes between my past updates (2.0.6.1 to 2.0.7.2 - bugfix) but I can't figure out that fast, was has caused that behavior.

So i might guess every command that is using this function should have problems.

Another thing that I haven't tested yet is setting HOME_AFTER_DEACTIVATE as work around.

sjasonsmith, 2020-11-23:
@C0unt23r0 `HOME_AFTER_DEACTIVATE` would not really be a workaround. Without that set, it is assumed that steppers can safely be disabled without losing position.

Given that you do not have this set. I do not think this is a bug. It is doing exactly what it is configured to do.

sjasonsmith, 2020-11-23:
To clarify, you _should_ set `HOME_AFTER_DEACTIVATE` if you plan to manually move your axis the way you described.

closed 2020-11-23 by sjasonsmith

C0unt23r0, 2020-11-23:
Well thanks @sjasonsmith. That makes it easy ;)

Setting HOME_AFTER_DEACTIVATE works as mentioned. Tested and verified.

The only thing i wonder is that i can not remember that the printer was behaving like this before updating and therefor i was thinking it might be a bug.