MarlinFirmware/Marlin #20237: [BUG] Dual X carriage - DANGEROUS MOVEMENT

DrumClock, 2020-11-21:
@InsanityAutomation 

### **ATTENTION!**

if **DUPLICATE** or **MIRRORED** mode is selected and code `G28` is received. 
Carriage X2 is parked, "Auto Home" is performed, and then carriage X2
strikes Endstop Xmax until the printer stops with the message `MSG_KILL_HOMING_FAILED
`

if the **AUTO-PARK** mode is selected from the LCD or the `M605 S1` command,
after executing `G28X`, the printer will restart

### **DANGEROUS MOVEMENT !**

In **AUTO-PARK** mode:
There is a carriage movement X2 between `X2_MIN` and Endstop Xmax
Code correction:
**Restrict X2 movement in the range `X2_MIN` to `X2_MAX`**

In **DUPLICATE** mode:
The X movement is in the range` X1_MIN `to `X1_MAX`
which causes a carriage X2 collision with the Endstop Xmax.
Code correction:
**restrict movement between `X1_MIN` and half the value of `X_BED_SIZE`**

In **MIRRORED** mode:
The X movement is in the range `X1_MIN` to `X1_MAX`
which causes carriage X2 to collide with X1.
Code correction:
**restrict movement in the `X1_MIN` range and add an `X_MAX_MIRRORED_POS`
setting that prevents both carriages from collapsing.**

[image]

DrumClock, 2020-12-08:
> if the **AUTO-PARK** mode is selected from the LCD or the `M605 S1` command,
> after executing `G28X`, the printer will restart

Hi @InsanityAutomation
if I use only `G28` instead of `G28X` after the command `M605 S1`, the printer will not be restarted.
**Please correct in the LCD menu - configuration of IDEX modes**

DrumClock, 2020-12-14:
HI @InsanityAutomation 
**Add behavior:**
when switching back to AUTO-PARK mode after command` M605 S1`:
`G28X` - reset printer
`G28XY` - reset printer
`G28XYZ` - change mode OK
`G28` - change mode OK

DrumClock, 2021-01-03:
HI @InsanityAutomation
Behavior without changes.
Tested on bugfix-2.0.x release 2021-01-03.

DrumClock, 2021-01-04:
> Hi @InsanityAutomation
> if I use only `G28` instead of `G28X` after the command `M605 S1`, the printer will not be restarted.
> **Please correct in the LCD menu - configuration of IDEX modes**

HI @InsanityAutomation
I confirm that after the merget bugfix https://github.com/MarlinFirmware/Marlin/pull/20654
The printer is not reset when the IDEX mode is changed to AUTO-PARK

Thans :)

Roxy-3D, 2021-02-12:
@DrumClock    Is this still a problem?   I just found this issue and saw it really wasn't getting the attention it deserves.

Can you give me the current status?

Thanks!

DrumClock, 2021-02-12:
@Roxy-3D Yes, unfortunately, no one solves it.

I gave up all efforts to beg.
As here https://github.com/MarlinFirmware/Marlin/issues/17941 and https://github.com/MarlinFirmware/Marlin/issues/19178

Roxy-3D, 2021-02-13:
I will say this:   You do a very good job of describing the issue and doing diagrams to help understand the issue!

I am very interested in IDEX issues and keeping things running.    I'm out of town for 6 more weeks and when I do get back to my machines I do not have as sophisticated of an IDEX setup as you do.   

With that said...   I'll help you work through these issues.   The IDEX code isn't that complicated.   But there ends up being a lot of corner cases.   

@InsanityAutomation   Can you take a quick look at #17941 and #19178  and see if there is anything simple that can be done to lessen @DrumClock 's pain?    And maybe suggest a strategy or direction for us to head towards resolution?

DrumClock, 2021-02-13:
> I am very interested in IDEX issues and keeping things running. I'm out of town for 6 more weeks and when I do get back to my machines I do not have as sophisticated of an IDEX setup as you do.

Hi @Roxy-3D 
I have the basic settings of IDEX mode.
 The https://github.com/MarlinFirmware/Marlin/issues/19178 not related to IDEX modem :)

I am testing IDEX on this test 3D frame of mine .... https://youtu.be/WkDNu3AFBKM

Roxy-3D, 2021-02-14:
That is a cool setup!   Now you need a heat bed and some hot ends so you can really print something with it.

DrumClock, 2021-02-14:
> That is a cool setup! Now you need a heat bed and some hot ends so you can really print something with it.

@Roxy-3D 
I use other 3D printers to print.
On this "frame" I test FW and new functions that come to my mind.
I don't really need to print to know that something is wrong.

InsanityAutomation, 2021-02-14:
On @17941 I know whats needed but havnt had time to sit down and do it. Its on my roadmap but I need to focus on what people sponsoring my work ask for first.

@19178 appears to be the same mesh boundary issue discussed over time on UBL, and I havnt gone through the configs to see if they have expanded the mesh over the full motion range as I do to avoid it.

DrumClock, 2021-06-06:
@InsanityAutomation 
je to tu od listopadu 2020 ..... mám to nechat otevřené ?

DrumClock, 2021-07-14:
@InsanityAutomation 
Because something doesn't work with each new release of bugfix 2.0.x 
(what used to be OK) and nobody solves it, **I decided to switch to FW Klipper.**
**I will no longer pay attention to these issues.**

shtaXveed, 2021-09-12:
anyone working on the endstop crashing yet?

whoishack, 2021-09-16:
> anyone working on the endstop crashing yet?
OP abandoned marlin for klipper, so there woudln't likely be any updates here. 

Take a look at this open issue, it's the same https://github.com/MarlinFirmware/Marlin/issues/22538

thisiskeithb, 2021-09-16:
Closing in favor of https://github.com/MarlinFirmware/Marlin/issues/22538

closed 2021-09-16 by thisiskeithb