MarlinFirmware/Marlin #20238: [BUG] Sensorless Homing only works in negative direction

bosd, 2020-11-21:
### Bug Description

Sensorless Homing only works in negative direction. When homing is done in postivie direction, the axis does not stop.

### Configuration Files

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5577954/Configuration.zip)

### Steps to Reproduce

1. Checkout latest bugfix 2.0 branch
2.  Install TMC2130 driver on both the X axis. (or any other stallguard supported driver)
3. Config marlin to use TMC2130 and Sensorless homing
4. `#define X_HOME_DIR -1`
5. #define `USE_XMAX_PLUG` and `USE_XMIN_PLUG`
6. Preform a Home verify your setup is correct (the home is in the minus direction)
7. Change X`_HOME_DIR -1` to `X_HOME_DIR 1`
8. Preform a Home

**Expected behavior:**
When homing in the positive direction, the axis should stop at the end of the travel

**Actual behavior:**

The axis keeps turning.

#### Additional Information

* Used board BTT GTR V1.0

bosd, 2020-11-22:
I’ve hooked up some endstops on Zmin and Zmax.
It appears that when homing in the - direction, it looks at the Zmax endstop.
On the GTR V1.0 board, the Stallguard is connected to the Zmin input.
Thus it will never trigger.
The inverting is the result of this code in endstops.h

`#define Z_ENDSTOP (Z_HOME_DIR < 0 ? TERN(HOMING_Z_WITH_PROBE, Z_MIN, Z_MIN_PROBE) : Z_MAX)`

As a workaround I could modify this code for my instance, or work with wire jumpers.

Not sure, if this issue is related. I do not want to hijack this thread, so please continue in #20238 .

boelle, 2020-11-22:
> On the GTR V1.0 board, the Stallguard is connected to the Zmin input.

if you cant change that then its a hardware error and not an error with marlin. error because board makers should allow people to decide what config

btw... is this printer with screws for Z or is it a delta that use belts?

bosd, 2020-11-22:
I would'nt count it an an hardware issue. As it is quite flexible how to setup the board.
(It is just a pin to the STM which happen to have an additional header to hookup and endstop as well)
That the `Z_HOME_DIR -1` parameter, changes the direction of the motor as well as the endstop pin is in this case not very helpfull.

Maybe the way stallguard is implemented in Marlin can be improved.
In such way the diag pin used for stallguard to be more flexible appointed.
(ex, parametrize it to endstop max, endstop min or probe)
As it can be used for: Travel Max, Travel Min (during homing or operation) and probing/object detection.

Would need to dig a bit deeper, to see if one can set the same pin for for the Max and Min endstop.

It's getting quite late here, so I'm not so sharp :)
Tried, multiple tests. Now on the Z-axis it is moving in the minus direction while homing. It stops on the Min endstop. But does not stop on the Sensorless homing.
(couple of days ago on the X, Y driver, it worked like a charm)

As debuging is quite complex, with pin assignments, inversions, postprocessing scipts ect.
Will get some sleep and do some more tests after some resting.

ellensp, 2020-11-23:
the stepper drivers only have one Diag pin. It is normally connected to the axis min endstop.
Most pins files will define a single endstop if STALL_SENSITIVITY is defined so sensorless will work as expected
Also depending on homing direction. It will define the unused port as min or max so it can be used.
Ie if homing direction is -1 the physical max port is defined as a max port, but if homing direction is 1 then the physical max port is defined as a min port so you can actually use it for something. (you cant use the physical min port as its used for sensorless)   
 
Looking at Marlin/src/pins/stm32f4/pins_BTT_GTR_V1_0.h it does not use this trick. 

Checking with the schematic confirms where the endstops and diags pins are. https://github.com/bigtreetech/BIGTREETECH-GTR-V1.0/blob/master/BTT%20GTR%20V1.0/manual/GTR%20V1.0(SCH).PDF

[pins_BTT_GTR_V1_0.zip](https://github.com/MarlinFirmware/Marlin/files/5581085/pins_BTT_GTR_V1_0.zip)

I have updated pins_BTT_GTR_V1_0.h with this trick.
Please try it

NB I cannot test this beyond compiling, Use caution.

boelle, 2020-11-23:
@bosd btw... is this printer with screws for Z or is it a delta that use belts?

bosd, 2020-11-25:
> @bosd btw... is this printer with screws for Z or is it a delta that use belts?

It is using screws for Z, belts for X Y

bosd, 2020-11-25:
Many Thanks to @ellensp !!
 The modded pins file is working fine for homing. In both directions. :+1: 

In my test setup the endstops are not physically connected to the screw. I trigger them by hand.
First attempt with the modded code resulted in some strange behaviour.
``Z_HOME_DIR -1``
I had to trigger the endstop by hand, only then the screw is stopped on the stallguard function.

The postprocessing did'nt like when I tried to change the endstop behaviour.
`Z_MIN_ENDSTOP_INVERTING false`
Throws an compiling error.
`#error "SENSORLESS_HOMING requires Z_MIN_ENDSTOP_INVERTING = true when homing to Z_MIN."`

Similarly when changing to ``Z_HOME_DIR 1`` the post processor throws an error:
`error: #error "Enable USE_ZMAX_PLUG when homing Z to MAX."`
These are some details which can be handled later on for a release.

With the modded code, homing works fine.
However, the screw can be driven to the end of the travel (and beyond).
When the axis is send to an position, the stallgaurd is ignored.
The motor tries to drive the leadscrew past it's end of travel, resulting in a lot of noise.

The modded code contains:
`#define Z_STOP_PIN                        Z_DIAG_PIN`
If I understand it correctly, this should make the Z-axis stop, as stallguard is triggerend on the Z_DIAG_PIN. For example when one trys to drive the screw it pasts it's travel. Or probing against a surface.
Is there some information how the `Z_STOP_PIN` is handled in Marlin.

ellensp, 2020-11-27:
endstops are only used in homing, they are not looked at during moves, unless you tell it to.

// If you want endstops to stay on (by default) even when not homing
// enable this option. Override at any time with M120, M121.
#define ENDSTOPS_ALWAYS_ON_DEFAUT

ellensp, 2020-11-27:
also sensor less is a really bad idea on z. and also for lead screws in general "axes with lead screws is absolutely not recommended and often doesn't work at all. The stall torque on the motor side is just too low and you risk damaging your bed."

bosd, 2020-11-27:
Thanks @ellensp 
I was aware of the endstop commands M120, M121. That function does work for the end stops. Not for the stallguard function.

Probably better, to use software endstops after homing.
(as I am unable to place physical endstops on this leadscrew.)

Well noted about the stall torque on leadscrews.
The machine is actually a cnc, which I actually want to apply torque with the Z-axis on the workpiece.
Was hoping to use sensorless probing with the leadscrew.

I don't think a belt system can provide enough force to push the tool against the workpiece.

sjasonsmith, 2020-11-28:
@bosd can you test the change @ellensp made, and close this issue if it is resolved?

sjasonsmith, 2020-11-28:
> The modded pins file is working fine for homing. In both directions. 

I didn’t see this further up in the comments. Based on this I will go ahead and close this issue. Let us know if that was incorrect!

closed 2020-11-28 by sjasonsmith