MarlinFirmware/Marlin #20245: Chamber temperature reading doesnt support other pins

italocjs, 2020-11-21:
This one is kinda weird and i didnt managed to find out what is wrong,

Version: Marlin 2.0.7.2 (from releases page)

resuming, my build has 2 nozzles, a heated bed and a heated chamber using a RAMPS1.4, so it needs T0, T1, T2 and something else

I was originally planning to use
T0 for nozzle 0 temp sensor
T1 for bed temp sensor
T2 for Nozzle 1 temp sensor 
AUX1 - PIN A3 (pin57) as chamber temp sensor 

For some reason i was getting random readings of this pin (between 19 to 30C, with or without a sensor attached), so i tried others (A4/D58 , A5/D59 and all other on aux pins) with the same results (except pin 63 which got me maxtemp error).

So i tried switching, chamber to pin 13 (T0, previously used for nozzle 0), and nozzle to pin 57 (A3)  and it somehow worked well.
I believe there is something in the code for reading the chamber temperature that is not properly reading or initializing other pins,  i'm attaching my full marlin folder (because i edited pins.h, config and config.adv0)|

I have tripe-checked this, and indeed it nozzle temp reading is able to read auxiliary pins, but not chamber,

[Marlin-2.0.7.2.zip](https://github.com/MarlinFirmware/Marlin/files/5578673/Marlin-2.0.7.2.zip)

sjasonsmith, 2020-11-21:
There is circuitry involved in a thermistor input, beyond just an analog input. Have you re-created that circuitry on your new pin?

italocjs, 2020-11-21:
Hi, thanks for the quick reply, i assembled the same circuit as used in ramps (10nf cap as filter + 4.7k pullup). but the readings are still weird, i'm checking agains another sensor and they are way off,

Is there somewhere else i should set a pin?  i mean like, on arduino IDE you need to set the pin type, if it has or not pullup.  i think something like this might be the issue

boelle, 2020-11-22:
> (10nf cap as filter + 4.7k pullup)

should be 10uF

[image]

italocjs, 2020-11-22:
> > (10nf cap as filter + 4.7k pullup)
> 
> should be 10uF
> 
> [image]

Sorry, wrote the wrong unit, but i used the right capacitor, following this same design.
It is working on 1.1.9,1 and not on 2.0.7.2, so i guess its not hardware related.

i tried checking the temperature routines, but it got quite complicated at a first glance

boelle, 2020-11-22:
Tried [`bugfix-2.0.x branch`](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) ?

boelle, 2020-11-22:
> arduino IDE

i dont think this will make any difference but you should switch to vscode+platform.io extension, many videos on how to install and setup

italocjs, 2020-11-22:
> > arduino IDE
> 
> i dont think this will make any difference but you should switch to vscode+platform.io extension, many videos on how to install and setup

I'm using vscode, i not even sure if its possible to compile marlin 2.0+ with arduino ide.  I did not tested bugfix, bug i have looked into the last few merges and none (as far as i seen) changed anything related to chamber.

boelle, 2020-11-22:
> compile marlin 2.0+ with arduino ide

you can but Arduino is so filled with bugs so its not worth it anymore

ellensp, 2020-11-22:
`#define TEMP_0_PIN                          57 
You should really use analog port name 3 as is the convention for analog pins. 

Just tried this with my own interface and your config (updated for bugfix) using Analog 3 as TEMP_0_PIN (with no thermistor and it reads -14 as expected)
Swapped TEMP_0_PIN and TEMP_CHAMBER_PIN, still reads -14 as expected
 
I suspect a wiring issue on your thermistor interface. 

Where did you get your vcc?  any chance you used the 5v rail on the servo plugs? As that is disconnected (floating) unless you connect vcc to 5v via the jumper (vcc and 5v pins on jumpers left of reset button) 

If not that perhaps you could attach a photo of how you have it wired.

italocjs, 2020-11-22:
I think i might be misundertanding how the pin are configured, 

When i set "logical pin 57", i am referring to the right pin D57/A3 on AUX 1)??  or should i use 3 or A3?
[image]

Please, which version did you use? i had success with 1.1.9.1, but not with 2.0.7.3. 

Aside from the termistor, i have added an extra output for steppers (dual Z) using pins 59, 64 and 44,  they worked without any issues.

Right now, my printer is working, using the chamber termistor on pin 65 (A11), and firmware version 1.1.9.1.

boelle, 2020-11-22:
that is digital pin 57 but analog pin 3

italocjs, 2020-11-22:
Bugfix from a few minutes ago, using port 12 (analog read 12 / pin 66 on ramps aux2) is working. i am not sure what changed, i searched through the changes but didnt found anything related, anyway, i guess its been fixed.

I would like to add a better chamber setting text later, specifying where to add the pins, temp sensor and how to build the circuit if needed.  this took me like two days (but i'm noob at programming)

closed 2020-11-22 by italocjs