MarlinFirmware/Marlin #20249: [BUG] Filament Change (M600) with cold Hotend (M302 P1) not working anymore

pseudex, 2020-11-22:
### Bug Description

The Filmanet Change (M600) does not allow a cold unload/load anymore.

### Configuration Files

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5579369/Configuration.zip)

### Steps to Reproduce

I have a Boden Printer with two Extruders. They join before the (single) hotend. The Filament is stored in front of the Y-Joiner after printing. If I want to swap the Filament before printing I do the following:

1. Send M302 P1 to allow Cold extrusion
2. Send M600 T1 to initiate Filament swap
3. Follow the Instructions on the Screen.

This was done on Marlin 2.0.5.3.
Now with Marilin 2.0.7.2 the M302 P1 command doesn't help to allow the cold extrusion anymore.

**Expected behavior:**

The Unload should ignore the temperature on the Hotend and do its thing.

**Actual behavior:**

Waits for 0°C to reach +-1°C (TEMP_WINDOW).
In this case I have three options. Put the printer in the fridge/outside, power off or send M108 (if e-parser enabled).

#### Additional Information

ensure_safe_temperature function in Pause.cpp is causing the issue.

```c
wait_for_heatup = true; // Allow interruption by Emergency Parser M108
  while (wait_for_heatup && ABS(thermalManager.degHotend(active_extruder) - thermalManager.degTargetHotend(active_extruder)) > TEMP_WINDOW)
    idle();
  wait_for_heatup = false;
```

this I changed to:

```c
  #if ENABLED(PREVENT_COLD_EXTRUSION)
    wait_for_heatup = !thermalManager.allow_cold_extrude;
  #else
    wait_for_heatup = true; // Allow interruption by Emergency Parser M108
  #endif
  while (wait_for_heatup && ABS(thermalManager.degHotend(active_extruder) - thermalManager.degTargetHotend(active_extruder)) > TEMP_WINDOW)
    idle();
  wait_for_heatup = false;
```
Here the commit on my fork of marlin: [AllowColdFilamentChange](https://gitea.kaiserflur.de/pseudex/Marlin_SKR1.3/commit/7988bb72a1741c2d44395f7599db040cb494297b)

Now I can swap the Filament again with a 0°C Hotend set temperature.
Without the M302 P1 command it will automatically heat the hotend to EXTRUDE_MINTEMP.
So in both ways I'm not stuck anymore to put my Printer in the fridge.

boelle, 2020-11-22:
> This was done on Marlin 2.0.5.3.
> Now with Marilin 2.0.7.2

Please test the [`bugfix-2.0.x branch`](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

pseudex, 2020-11-22:
I did a quick compare of the function **ensure_safe_temperature**. In V2.0.7.2 and the actual bugfix 2.0.x it is exactly the same.

qwewer0, 2020-11-22:
> I did a quick compare of the function **ensure_safe_temperature**. In V2.0.7.2 and the actual bugfix 2.0.x it is exactly the same.

Didn't meant to compare the code, but to actually test it.

I tested `Change Filament` with `M302 P1` on the latest Bugfix-2.0.x and it indefinitely waits for the nozzle to be heated up to 0°C.

qwewer0, 2020-11-28:
@pseudex https://github.com/MarlinFirmware/Marlin/pull/20262 has been merged, could you close this if it fixed the issue?

closed 2020-11-29 by pseudex