MarlinFirmware/Marlin #20250: Suggestion for auto-loading bed mesh data based on target temperature

TanmayChhatbar, 2020-11-22:
Been trying to use UBL, and I find that I have to manually load the mesh that I want at every boot-up, and re-load another mesh if I want to print at a different temperature. It would be handy if there was an auto-loading option for +-5 degrees (or closest) of the temperature at which the mesh was generated. Do let me know if I'm not using it right though. Thanks.

ManuelMcLure, 2020-11-22:
This is easily handled in your slicer start gcode. For example, PrusaSlicer/SuperSlicer have filament-specific start gcode and it's very easy to just add a `M420 L` command that matches the temperature you are going to be using.

boelle, 2020-11-22:
this is more a support Q and that is not what the issue list is for so will close this one

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-11-22 by boelle