MarlinFirmware/Marlin #20252: [BUG] S curve exceeds maximal acceleration

espr14, 2020-11-22:
### Bug Description

S curve causes step skipping. The same G code without S curve prints great.

### Configuration Files

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5579498/Marlin.zip)

### Steps to Reproduce

1. Use FW with S curve on.
1. Set the printer to handle acceleration just 3000 mm/s^2 in X axis.
1. Run the G code.

**Expected behavior:**

Should not loose the steps.

**Actual behavior:**

Loose steps.

#### Additional Information

* I haven't seen this with 2.0.x (with different G codes)
* Does not happen with disabled S curve

[S_curve.zip](https://github.com/MarlinFirmware/Marlin/files/5579513/S_curve.zip)

boelle, 2020-11-22:
> * I haven't seen this with 2.0.x

so this does not happen with [`bugfix-2.0.x branch`](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) ?

espr14, 2020-11-22:
bugfix-2.0.x: does
2.0.x: very likely not

rhapsodyv, 2020-11-30:
Is it the same as #20277 ?

espr14, 2020-11-30:
I don't know if it's the same but behaves the same. However, with junction deviation enabled it looses steps more and with S curve it looses steps even more.

closed 2021-02-09 by github-actions[bot]