MarlinFirmware/Marlin #20253: PAT9125 Filament Sensor

Guga1310, 2020-11-22:
I would like to use the PAT9125 filament sensor on the i2c input of the SKR 1.4 Turbo board

sjasonsmith, 2020-11-22:
There is really not enough information here to warrant keeping this open. There is a template for feature requests which was ignored here.

My immediate questions are:

1. How does somebody obtain a PAT9125 sensor?
   - Is it available to purchase pre-built?
   - Are plans available to build your own?
2. How does your request differ from existing Marlin support?
3. What new value does this capability bring to Marlin?

LinFor, 2020-11-23:
There have already been requests to use this sensor, there were even some attempts to implement its support: #8062, #12962...

ellensp, 2020-11-23:
This sensor is unreliable, even Prusa dropped it like a hot rock very quickly.

Guga1310, 2020-11-23:
> Já houve pedidos para usar este sensor, houve até algumas tentativas de implementar o seu suporte: [# 8062](https://github.com/MarlinFirmware/Marlin/issues/8062) , [# 12962](https://github.com/MarlinFirmware/Marlin/pull/12962) ...

Thanks

Guga1310, 2020-11-23:
> Este sensor não é confiável, até Prusa o deixou cair como uma rocha quente muito rapidamente.

Thanks

Guga1310, 2020-11-23:
> Na verdade, não há informações suficientes aqui para garantir que isso seja mantido em aberto. Há um modelo para solicitações de recursos que foi ignorado aqui.
> 
> Minhas perguntas imediatas são:
> 
> 1. Como alguém obtém um sensor PAT9125?
>    
>    * Ele está disponível para compra pré-fabricado?
>    * Existem planos para construir o seu próprio?
> 2. Qual a diferença entre o seu pedido e o suporte existente do Marlin?
> 3. Que novo valor esse recurso traz para a Marlin?

Thanks

closed 2020-11-23 by Guga1310