MarlinFirmware/Marlin #20255: [FR] Binary format for gcode-files for faster transfer

jtkDvlp, 2020-11-23:
### Description

Hi there,

a few weeks ago I decided to write my own wifi printer client (not public yet, will comming). Its working well, I also upload gcode files via wifi to the printers sd card and of course as others already mentioned its slow caused by the serial interface. Of course you could try to get the communication better, but I thinks thats difficult if you dont want to make big changes to the communication itself. But the gcode transfered actual is human readable. Of course thats useful if you want to control the printer via console. But for printing you dont do this just for configuring. So there is no need for human readable gcode files per se. So I decied to give it a try and two afternoon later I had written a [snippet converting gcode to binary gcode and vise versa saving up to 50% file size](https://gist.github.com/jtkDvlp/05322a3c333410f1b89ce98808964d9d). For my simplicity its written in [clojure](https://wikipedia.org/wiki/Clojure). But I also could provide an JavaScript Version for clients and C / C++ Version for the firmware.

If there is interest in such a feature and the marlin contributors are willing to accept such a pull request, it would be nice to get this done. As already mentioned I can do / help with this.

For writting the C / C++ part for the firmware it would be nice to get a little api description I have to provide or some direct source links to read getting an overview for the integration.

Best regards
jtkdvlp

#### Additional Information

Clojure snippet to convert gcode to binary gcode and vise versa saving up to 50% file size. 
https://gist.github.com/jtkDvlp/05322a3c333410f1b89ce98808964d9d

ellensp, 2020-11-23:
There have been several attempts at this
see BINARY_FILE_TRANSFER (currently in Marlin) 
and https://reprap.org/wiki/Packed_Binary_GCode

jtkDvlp, 2020-11-23:
Mmh XD perfect!

 So this option accepts gcode files with schema of the wiki articel? Do you know some client code for this? Otherwise I tweak my code.

ellensp, 2020-11-23:
Different systems.

From reading the code BINARY_FILE_TRANSFER uses compression, this https://github.com/atomicobject/heatshrink

rhapsodyv, 2020-11-23:
Here is the original PR. It have a script for upload binary data to marlin too. https://github.com/MarlinFirmware/Marlin/pull/14817

jtkDvlp, 2020-11-23:
@ellensp  does BINARY_FILE_TRANSFER work. I send `M28 B1 foo`, then sending some binary stuff and ending with `M29 foo` but the printer does not respond. Any ideas?

p3p, 2020-11-23:
`BINARY_FILE_TRANSFER` #14817 as it is currently in Marlin, can only be used to transfer files to the sdcard. To use `BINARY_FILE_TRANSFER` you need to implement the protocol as described, or just use the script provided, I think there's an octoprint plugin that takes advantage of it for GCode uploads. 

It is still a project I'm working on and the version in Marlin is unfinished (my dev version is nothing like the it), other things have taken priority, at its core this feature is just a multi channel serial binary transport layer, and you could implement any binary protocol you wanted on top of it, I think streaming compressed gcode is enough of a bandwidth improvement, but you could also implement a binary gcode protocol which would probably be more computationally efficient on the mcu.

ellensp, 2020-11-24:
Never looked at it. With modern 32 bit controllers with CDC serials ports (ie serial at near usb2 speeds, regardless of baud rate) 
It is rather "unfashionable"

But I just gave it a try.

NB this is is test mode:
transfer.py -p /dev/ttyACM0 -b 250000 -t 3DBenchy.gcode 
results where 
Connecting: Switching Marlin to Binary Protocol...
Connection synced [0], binary protocol version 0.1.0, 96 byte payload buffer
File Transfer version: 0.1.0, compression: heatshrink
3DBenchy.gcode opened
100.00% 7.26KiB/s  Errors: 1
then transfer errored...

With Compression
transfer.py -p /dev/ttyACM0 -b 250000 -t -c 3DBenchy.gcode 
onnecting: Switching Marlin to Binary Protocol...
Connection synced [168], binary protocol version 0.1.0, 96 byte payload buffer
File Transfer version: 0.1.0, compression: heatshrink
3DBenchy_0.2mm_ABS_MK2.5_1h35m.gcode opened
100.00% 7.25KiB/s [16.50KiB/s] Errors: 20
then transfer errored...

Error is "MarlinBinaryProtocol.py", line 316, in await_response
    raise ReadTimeout()
MarlinBinaryProtocol.ReadTimeout
"

When I try it for real compresses or just bin. it errors with
```
MarlinBinaryProtocol.py", line 357, in open
    raise Exception("Can not open file on client")
Exception: Can not open file on client

```

jtkDvlp, 2020-11-24:
@p3p do you have slack or discord for this feature?

p3p, 2020-11-24:
Nothing specific to the feature no, I'm on the Marlin discord if you would like to discus something.