MarlinFirmware/Marlin #20261: [BUG] ENDSTOP_INTERRUPTS_FEATURE makes hotend crash whilst hominig

DJScopeSOFM, 2020-11-24:
### Bug Description

ENDSTOP_INTERRUPTS_FEATURE makes hotend crash whilst homing.

### Steps to Reproduce

As above

**Expected behavior:**

Homing as normal

**Actual behavior:**

Crash

#### Additional Information

Current latest Bugfix as of 24/11/2020. 
MKS Robin Nano 1.2 - Two Trees Sapphire Plus

LinFor, 2020-11-24:
What are you mean "crash"?
Does all endstop pins of your board are interrupts-capable?

cohaolain, 2020-11-24:
Hey, quick question - is this happening **every** time you home, or is it happening **intermittently**?

DJScopeSOFM, 2020-11-24:
> What are you mean "crash"?
> Does all endstop pins of your board are interrupts-capable?

Carriage continues to to grind into endstop. Yes, they're capable. This only started happening after updating to the latest bugfix sources.

DJScopeSOFM, 2020-11-24:
> Hey, quick question - is this happening **every** time you home, or is it happening **intermittently**?

Every single time until I comment it out. I've tried reducing homing speed and increasing bump distance. But it will still happen until I comment it out.

cohaolain, 2020-11-24:
Ah, interesting. I had thought it might be related to issue #20082 (where the z interrupt call doesn't seem to complete) where similar behaviour occurs intermittently, sorry!

DJScopeSOFM, 2020-11-24:
> Ah, interesting. I had thought it might be related to issue #20082 (where the z interrupt call doesn't seem to complete) where similar behaviour occurs intermittently, sorry!

I think the endstop interrupt feature is broken entirely at the moment.

boelle, 2020-11-24:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

DJScopeSOFM, 2020-11-30:
> Configurations, please
> Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
> requested in the Issue template) and drop them into your next reply.
> We'll check them over and see if anything is amiss.

Apologies for not responding sooner. I've changed the files quite a few time since posting this. You can find the files on my GitHub: https://github.com/DJScopeSOFM/Two-Trees-Multi-Printer-Marlin-Bugfix-Compiler

That was the only option that caused this behaviour and removing it, the behaviour stopped. I don't know if it's isolated or not, other users using the same printer have said that they had the same behaviour.

closed 2021-01-07 by github-actions[bot]