MarlinFirmware/Marlin #20264: [BUG] Software endstop fails after M84

Sergey1560, 2020-11-24:
With the HOME_AFTER_DEACTIVATE option enabled, software endstops do not work when moving off the bed. For example maximum for Y-axis is 200. 

```
G28 ;home 
G1 Y220 ; Y stop at 200, software end stop works
M84 ;Disable axis
G1 Y220 ; Y trying to go to 220
```
This is because in the Marlin/src/module/motion.cpp file, "apply_motion_limits" function check limits only if the axis is homed.
```
    if (TEST(axis_homed, Y_AXIS)) {
        #if !HAS_SOFTWARE_ENDSTOPS || ENABLED(MIN_SOFTWARE_ENDSTOP_Y)
          NOLESS(target.y, soft_endstop.min.y);
        #endif
        #if !HAS_SOFTWARE_ENDSTOPS || ENABLED(MAX_SOFTWARE_ENDSTOP_Y)
          NOMORE(target.y, soft_endstop.max.y);
        #endif
      }
```

Expected behavior: 
Do not move without homing when the option HOME_AFTER_DEACTIVATE is enabled or take into account in some way software endstop.

I checked the behavior on the latest bugfix branch, the problem persists.

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5588328/Configuration.zip)

swissnorp, 2020-11-26:
Well after thinking a bit more about the issue and the pull request, I must say that this is not a bug and the behaviour is absolutely okey. We may just need to have a note, that HOME_AFTER_DEACTIVATE does not prevent movement!

NO_MOTION_BEFORE_HOMING: prevents moving if axes not homed, thats the case!
HOME_AFTER_DEACTIVATE: forces axes to be rehomed after deactivated steppers, thats the case too!

If you want to prevent movement after steppers have been disabled, you need to enable both of them.

Sergey1560, 2020-11-26:
> Well after thinking a bit more about the issue and the pull request, I must say that this is not a bug and the behaviour is absolutely okey. We may just need to have a note, that [...]

The problem is not that the movement is not prohibited, but that the software endstops are not taken into account. In case of errors in the gcode, this can lead to damage.

swissnorp, 2020-11-26:
I know, I know...
If `NO_MOTION_BEFORE_HOMING` is disabled and you turn on your printer, no axis is homed but you can move each axis. It makes no sense to have soft endstops if the positions on the axes are not homed. This is the same state as with `HOME_AFTER_DEACTIVATE` enabled but `NO_MOTION_BEFORE_HOMING` disabled.

Sergey1560, 2020-11-27:
Fixed by #20283

closed 2020-11-27 by Sergey1560