MarlinFirmware/Marlin #20265: [BUG] RADDS speaker/buzzer always enunciating

ej0rge, 2020-11-24:
### Bug Description

When a control panel with a buzzer is attached to the buzzer pin (41) on J3 of the RADDS, it is always making noise once marlin has booted. Using adapter cable based on radds2lcd project and either rrd full graphic or fysetc mini 12864.

### Configuration Files

[x5s-radds.zip](https://github.com/MarlinFirmware/Marlin/files/5591508/x5s-radds.zip)

### Steps to Reproduce

1. Connect beeper wire of control panel to the corresponding pin on radds
2. power on and boot up marlin

**Expected behavior:**

regular marlin type beeps and chirps

**Actual behavior:**

"Eeeeeeeeeeeeeeeee" which never ends. M300 commands do not result in any change in tone or break in continual sound.

#### Additional Information

It's always done this since before marlin 2 was released but it hadn't risen into the top 10 issues i was annoyed by. I was also less certain before recently that i had connected the display 100% correct according to the designs left by those who came before me.

ellensp, 2020-11-24:
"a buzzer is attached to the buzzer pin (41)"  but yet you have #define SPEAKER

ej0rge, 2020-11-24:
> "a buzzer is attached to the buzzer pin (41)" but yet you have #define SPEAKER

If you look at the attached configuration, I did #define SPEAKER.

ej0rge, 2020-11-24:
[RADDS_27_sch.pdf](https://github.com/MarlinFirmware/Marlin/files/5593213/RADDS_27_sch.pdf)

I found the 1.6 version of the radds schematic and it seems to show that pin 41 is brought directly to the pin on the J3 header. 

The schematic for my Fysetc mini 12864 v1.2 indicates that the 'buzzer' is a KSSGJ4B20 

Fysetc schematic here: 
https://github.com/FYSETC/FYSETC-Mini-12864-Panel/blob/master/V1.2/mini12864%E5%8E%9F%E7%90%86%E5%9B%BE.pdf

Datasheet for KSSGJ4B20 does call it a 'buzzer' but includes a frequency response graph and indicates that it is rated to receive a 2730hz square wave. This strongly suggests to me that defining SPEAKER is justified. 

http://www.farnell.com/datasheets/97523.pdf

ej0rge, 2020-12-27:
This appears to have been fixed by replacing my sketchy chinese knockoff DUE with an Arduino Genuino DUE.

closed 2021-02-05 by github-actions[bot]