MarlinFirmware/Marlin #20267: Latest bugfix doesn't work on Anycubuc Predator

rq3, 2020-11-24:
# NO SUPPORT REQUESTS PLEASE

Do you want to ask a question? Are you looking for support? Please don't post here. Support Requests posted here will be automatically closed!

Instead use one of the following options:

- The Marlin Firmware forum at https://reprap.org/forum/list.php?415
- The MarlinFirmware Facebook Group at https://www.facebook.com/groups/1049718498464482/
- The MarlinFirmware Discord Server at https://discord.gg/n5NJ59y.

Before filing an issue be sure to test the latest "bugfix" branch to see whether the issue is already addressed.

rq3, 2020-11-24:
The latest bugfix, (11/24/20), using the latest example configuration and configuration_adv files does not work for the Anycubic Predator. It compiles and loads, but the screen colors are wrong, and there is no touch functionality at all.

There is no point in attaching the files, since they are the latest files on the Marlin bugfix.

sjasonsmith, 2020-11-24:
Try using the example from the import-2.0.x branch. This example has been updated recently and the latest changes may not be in other branches.

ellensp, 2020-11-25:
you say "There is no point in attaching the files,"  I say where?
https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.0.x/config/examples/AnyCubic only lists 4 machines the Predator
 is not on of them.
Also bugfix itself comes with a stock ramps, not configured for any particular machine

ellensp, 2020-11-25:
Ah its a delta! https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.0.x/config/examples/delta/Anycubic/Predator
Wouldn't have been an issue if you had included Configs!.
They are ALWAYS needed.

rhapsodyv, 2020-11-25:
> It compiles and loads, but the screen colors are wrong, and there is no touch functionality at all.

What do it means? Can you provide photo/video?

For touch, did you tried send M995 to calibrate the touch?

boelle, 2020-11-25:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

Please test the [`bugfix-2.0.x branch`](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

sjasonsmith, 2020-11-28:
I am closing this, since we have other indications that this examples is working. Without this report following template instructions I don't think it is worth any time to pursue any changes.

Just 10 days ago the example configuration was updated to fix screen rotation. You should certainly test using the examples from the `import-2.0.x` branch in the configurations repo, to be sure you are using the latest examples.
https://github.com/MarlinFirmware/Configurations/pull/328

If there are still issues, please be sure future bug reports follow the issue template and provide all the requested information. I a case such as this one I would expect pictures of the misbehaving screen as well.

closed 2020-11-28 by sjasonsmith