MarlinFirmware/Marlin #20271: MKS Robin Nano v1.1. Display icons update interruption

altiber, 2020-11-25:
### Bug Description

MKS Robin Nano v 1.1 . The update of icons on the TFT display with the #define TFT_LG_UI mode enabled and 'assets'copied to the SD card is interrupted. After that, instead of icons, there are white squares

### Configuration Files

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5594783/Marlin.zip)

### Steps to Reproduce

1. Build Marlin
2. Copy firmware and Assets folder to SD card
3. Got problev

**Expected behavior:**

Display with icons after firmware update

rhapsodyv, 2020-11-25:
Are you using bugfix?

Can you format the SD card and try again?

altiber, 2020-11-25:
I formatted the sd card, tried another one. Without result. Yes, I used bugfix, tried it for a few days. While using https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/tree/RogueM-pull-request-more-menu everything is fine, the icons are updated.

rhapsodyv, 2020-12-03:
Recently an user on MKS Facebook group had a similar issue. The problem was the SD card greater than 8gb... 

Did test it in a smaller SD card? Just to check...

altiber, 2020-12-04:
I checked on 8 and 16 GB cards, and there was definitely a problem with them. I do not have less volume cards, and it is difficult to buy them, they are no longer sold in stores in our city

rhapsodyv, 2020-12-07:
I just tested it in my Nano V2 and my Nano V1.2 and it worked 100%. I used a 512mb card. Can you test bugfix again?

I will try ask MKS what is the diference between nano v1.1 and nano v1.2. 

What is the name of the firmware file you are uploading? If you are using Robin_nano35.bin, try updating Robin_nano.bin (just once, because it could reset some offset settings).

mks-viva, 2020-12-08:
MKS Robin Nano V1.2（Modified based on V1.1）
Add 1 normally open FAN 
Increase 3D TOUCH interface, use PA8
WIFI control pin adjusted to PC13
TF card, USB port protruding from the edge of the board 2mm, USB power supply is optional

altiber, 2020-12-08:
Ok, i'l try less volume sd card, but it will take few days to get it.  My fw file was robin_nano35.bin.

MKS-Sean, 2020-12-08:
[image]
Use the latest marlin-bugfix version, you can update to the assets file

altiber, 2020-12-09:
I confirm that everything works in the latest version of the bugfix. Still, it was not a false alarm, but a correction over time of issue ;).

boelle, 2020-12-09:
@altiber if  everything works and your initial bug is gone please click the green close button below

closed 2020-12-09 by altiber

rhapsodyv, 2020-12-09:
> I confirm that everything works in the latest version of the bugfix. Still, it was not a false alarm, but a correction over time of issue ;).

Yes. Bugfix change a few times a day, so sometimes things could be broken for a few days.

Thanks!