MarlinFirmware/Marlin #20272: [BUG] Ender 3 v2 not booting with SDCARD_EEPROM_EMULATION

ct16k, 2020-11-25:
### Bug Description

I updated from stock firmware to latest release, since I wanted to add a BLTouch to my Ender 3 v2. But I wanted to retain the functionality of saving settings to SD card, as the old firmware did, so I defined `SDCARD_EEPROM_EMULATION` in Configuration.h.

The updated firmware seems to flash fine and an `EEPROM.DAT` file is created on the card, but the printer never gets past the bootup screen.

I tried latest bugfix-2.0.x (2693e35caead85de734735c852de52bebbdc818b at the time) to no avail.

### Configuration Files

[config.zip](https://github.com/MarlinFirmware/Marlin/files/5594858/config.zip)
I included the generated `EEPROM.DAT` file as well if it helps.

### Steps to Reproduce

1. Build latest release or bugfix-2.0.x fw with SDCARD_EEPROM_EMULATION enabled
2. Flash it

**Expected behavior:**

Printer boots up properly, saves settings to eeprom.dat instead of EEPROM.

**Actual behavior:**

Stuck on boot screen with progress bar looping indefinitely.

#### Additional Information

I also enabled `SD_CHECK_AND_RETRY` and thought it might have something to do with it, but disabling it doesn't help.

boelle, 2020-11-25:
did you use these configs? https://github.com/MarlinFirmware/Configurations/tree/import-2.0.x/config/examples/Creality/Ender-3%20V2

ct16k, 2020-11-25:
For the release build I used the configs from the [release-2.0.7.2 branch](https://github.com/MarlinFirmware/Configurations/tree/release-2.0.7.2/config/examples/Creality/Ender-3%20V2), and for the `bugfix-2.0.x` build I used configs from the [bugfix-2.0.x branch](https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.0.x/config/examples/Creality/Ender-3%20V2).

According to git there is no difference between `bugfix-2.0.x` and `import-2.0.x`, except for the extra `CONFIG_EXAMPLES_DIR` macro in the former:

```
>git diff import-2.0.x -- "config/examples/Creality/Ender-3 V2"
diff --git a/config/examples/Creality/Ender-3 V2/Configuration.h b/config/examples/Creality/Ender-3 V2/Configuration.h
index 3f4fbd296..2caa44271 100644
--- a/config/examples/Creality/Ender-3 V2/Configuration.h
+++ b/config/examples/Creality/Ender-3 V2/Configuration.h
@@ -21,6 +21,8 @@
  */
 #pragma once

+#define CONFIG_EXAMPLES_DIR "Creality/Ender-3 V2"
+
 /**
  * Configuration.h
  *
diff --git a/config/examples/Creality/Ender-3 V2/Configuration_adv.h b/config/examples/Creality/Ender-3 V2/Configuration_adv.h
index 34f7e4ae8..76ad6f89a 100644
--- a/config/examples/Creality/Ender-3 V2/Configuration_adv.h
+++ b/config/examples/Creality/Ender-3 V2/Configuration_adv.h
@@ -21,6 +21,8 @@
  */
 #pragma once

+#define CONFIG_EXAMPLES_DIR "Creality/Ender-3 V2"
+
 /**
  * Configuration_adv.h
  *
diff --git a/config/examples/Creality/Ender-3 V2/_Bootscreen.h b/config/examples/Creality/Ender-3 V2/_Bootscreen.h
index 324cb4a61..cfa856dec 100644
--- a/config/examples/Creality/Ender-3 V2/_Bootscreen.h
+++ b/config/examples/Creality/Ender-3 V2/_Bootscreen.h
@@ -21,6 +21,8 @@
  */
 #pragma once

+#define CONFIG_EXAMPLES_DIR "Creality/Ender-3 V2"
+
 /**
  * Custom Boot Screen bitmap
  *
diff --git a/config/examples/Creality/Ender-3 V2/_Statusscreen.h b/config/examples/Creality/Ender-3 V2/_Statusscreen.h
index d1e2f4a64..60e79759c 100644
--- a/config/examples/Creality/Ender-3 V2/_Statusscreen.h
+++ b/config/examples/Creality/Ender-3 V2/_Statusscreen.h
@@ -21,6 +21,8 @@
  */
 #pragma once

+#define CONFIG_EXAMPLES_DIR "Creality/Ender-3 V2"
+
 /**
  * Custom Status Screen bitmap
  *
```

sjasonsmith, 2020-12-02:
Is there any value in spending time on this? This board has a built-in EEPROM, so SDCARD_EEPROM_EMULATION is not necessary.

thisiskeithb, 2020-12-02:
> Is there any value in spending time on this? This board has a built-in EEPROM, so SDCARD_EEPROM_EMULATION is not necessary.

We could just strip out the other options and default to the real EEPROM like we've done on other boards (GTR, SKR Mini E3 V2, E3 Turbo, etc.)

ct16k, 2020-12-02:
The example config already defaults to using EEPROM. Enabling emulation was a customization of my config.

Without an option to manually back up the EEPROM, having it permanently on the SD card is a good workaround to be able to back it up when switching firmwares instead of having to re-enter values every time. But I don't have strong feelings about this. I'm not tweaking that many params anyway, so it's not that big of a chore, just more convenient.

If this is not going to be fixed, maybe amend the sanity checks to not allow `SDCARD_EEPROM_EMULATION` for the Ender 3 V2 with a Creality V4.2.2 board?

sjasonsmith, 2020-12-02:
@ct16k, the changes in the above-reference pull request should fix it for you. You may specify `SDCARD_EEPROM_EMULATION` anywhere in your `Configuration.h` file, or edit the pins file to enable it in place of the IIC EEPROM.

ct16k, 2020-12-02:
Printer boots up, changes seem to be persisted to file and load up fine, so I would say it does fix it. Thank you!

PR #20353 (2020-12-02): Fix SDCARD_EEPROM_EMULATION hang on E3V2
### Description

The E3V2 DWIN display code assumes it is using `IIC_BL24CXX_EEPROM` and tries to access it directly to store the screen's language setting. This caused hangs if users tried to override this and use SDCARD_EEPROM_EMULATION, and the EEPROM was accessed anyway to retrieve the screen's language setting.

This change resolves issues surrounding SDCARD_EEPROM_EMULATION, although the screen's language setting will not be stored.

### Benefits

Can use SD EEPROM if you really want to.

### Configurations

See included automated test change for  [... cut]
closed 2020-12-02 by sjasonsmith