MarlinFirmware/Marlin #20273: MKS RUMBA32 use MKS OLED no working

mks-viva, 2020-11-25:
### Bug Description

- Use MKS RUMBA32 motherboard, connect MKS OLED0.96 display, but no working and i has add pin define to pins_RUMBA32_common.h file
  ```
    #if EITHER(MKS_12864OLED, MKS_12864OLED_SSD1306)
      #define LCD_PINS_DC                     PE13  // Set as output on init
      #define LCD_PINS_RS                     PE14  // Pull low for 1s to init
      // DOGM SPI LCD Support
      #define DOGLCD_A0                       LCD_PINS_DC
      #define DOGLCD_CS                       PE10
      #define DOGLCD_MOSI                     PE9
      #define DOGLCD_SCK                      PE12
    #endif
  ```
- Use 2560 motherboard and MKS OLED0.96 will working
- I am run offical marlin bug-fix 2.0.x version

### Configuration Files

[Config and pin.zip](https://github.com/MarlinFirmware/Marlin/files/5595821/Config.and.pin.zip)

### Issues : https://github.com/makerbase-mks/MKS-RUMBA32/issues/27

alex18881, 2020-11-25:
Also I can confirm exactly the same issue with BIGTREETECH SKR mini V1.1. Here's my configs:
[conf.zip](https://github.com/MarlinFirmware/Marlin/files/5596038/conf.zip)

sjasonsmith, 2020-11-28:
With the huge variety of displays available it is extremely difficult to have every display configured properly for every board. It would be most helpful if people using this hardware can contribute pull requests to help resolve the incompatibilities.

closed 2021-01-05 by github-actions[bot]