MarlinFirmware/Marlin #20274: [BUG] Items in `Control` menu get shifted down by one when going to `Info` screen

ct16k, 2020-11-25:
### Bug Description

I updated from stock firmware to latest release, since I wanted to add a BLTouch to my Ender 3 v2. After successfully building a `release-2.0.7.2` firmware with BLTouch enabled, I updated both the firmware on the printer, as well as that on the LCD, using the [example files](https://github.com/MarlinFirmware/Configurations/tree/release-2.0.7.2/config/examples/Creality/Ender-3%20V2/DWIN_SET) provided. I now have a `Level` option on the main screen, instead of `Info`, which has moved under `Control`.

If I browse to the `Info` screen and back, all items under control are now rendered one line lower than they should be, with the top line empty and the `Info` option just above the nozzle temp indicators. This `Info` text remains there until the printer is shut down.

I also tried latest bugfix-2.0.x (2693e35caead85de734735c852de52bebbdc818b at the time), but it didn't make a difference.

### Configuration Files

[config.zip](https://github.com/MarlinFirmware/Marlin/files/5594858/config.zip)
Archive contains an unrelated `EEPROM.DAT` file which can be ignored (it's there for a separate issue).

### Steps to Reproduce

1. Build latest release or bugfix-2.0.x fw with leveling options
2. Flash printer and LCD
3. Go into Control/Info and back

**Expected behavior:**

The usual Control menu

**Actual behavior:**

All items shifted down, with `Info` item remaining on the display afterwards until printer is powered off.

#### Additional Information

[image]
[image]

ellensp, 2020-11-25:
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

ct16k, 2020-11-25:
As mentioned in the report above, I've already tried `bugfix-2.0.x`. There has been only one commit since I then (04c4c6004e752582235357b14aebdd5ff579f477), which relates to touch screens, so not applicable to this off-by-one error.

ellensp, 2020-11-25:
Opps, missed that line. Sorry

sjasonsmith, 2020-12-01:
I can confirm this behavior. When going back the icons all appear correctly, and all the labels are shifted one down (with the last label dropped completely). Selecting an item performs the action represented by the icon.

sjasonsmith, 2020-12-01:
This issue was very easy to reproduce and test, so I have gone ahead and merged a fix. I'm closing this issue. Please reply if you have problems with my fix, or report new issues if more menu problems are found.

closed 2020-12-01 by sjasonsmith

ct16k, 2020-12-01:
Can also confirm this is fixed. Thank you!

PR #20338 (2020-12-01): Fix E3V2 Control Menu when returning from Info
### Description

The Control menu did not draw properly after viewing Info. This was due to not factoring in the scrolled position when drawing the labels.

A macro already existed to account for scrolling, but was only used with Chinese. This change also applies it to English labels.

Refer to #20274 for more information and pictures.

### Benefits

Coherent screen display.

### Configurations

Ender 3 V2 example, modified with BLTouch, 4.2.7 board, etc.
[Configs.zip](https://github.com/MarlinFirmware/Marlin/files/5621170/Configs.zip)

### Relate [... cut]