MarlinFirmware/Marlin #20275: [BUG] Disabling SEGMENT_LEVELED_MOVES Causes Board to Reset - Mesh Bed Leveling (Manual without probe)

houseofbugs, 2020-11-25:
Tested with Bugfix and 2.0.7.2.

When enabling the manual mesh bed leveling and having SEGMENT_LEVELED_MOVES disabled this causes a reset of the printer board. Happens on multiple boards, not board specific.

vitaliy172, 2020-11-25:
Please, follow Issue template and include your configs

ellensp, 2020-11-25:
Please follow the Issue Template when filing a bug report. This includes attaching your configs:
 
```lisp
<!--

Have you read Marlin's Code of Conduct? By filing an Issue, you are expected to comply with it, including treating everyone with respect: https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/.github/code_of_conduct.md

Do you want to ask a question? Are you looking for support? Please don't post here. Instead please use the Marlin Firmware forum at http://forums.reprap.org/list.php?415 or the Marlin Facebook Group https://www.facebook.com/groups/1049718498464482/ or the Marlin Discord Server https://discord.gg/n5NJ59y.

Before filing an issue be sure to test the 1.1 and/or 2.0 "bugfix" branches to see whether the issue is already addressed.

-->

### Description

<!-- Description of the bug or requested feature -->

### Steps to Reproduce

<!-- If this is a Bug Report, please describe the steps needed to reproduce the issue -->

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.
```

houseofbugs, 2020-11-28:
It's not config specific and affects 2.0.7.2 and bugfix. But if you want to not be aware of the bug then that's fine, I'll stop reporting issues and just keep fixing things on my own branches.

ellensp, 2020-11-28:
Config files provide invaluable information required. Like what type of motherboard? And any other stetting that may come into play.
If you have the ability to fix this issue, you should provide a PR to do so, not post a "I have a bug" with zero details

houseofbugs, 2020-11-28:
I updated with the *exact* steps to reproduce. I have not had time to fix the issue but just enabled the EGMENT_LEVELED_MOVES option until I can find out why. I was reporting it because there is a workaround but I don't have time to figure out why disabling it causes the crash. It is not board specific like I said. Happens on Creality 2560 Boards, 1284p Creality boards, MKS SGENL V2, EZBoard Lite, and SKR E3 Mini (all versions). I tested this on all these boards before making the report.

sjasonsmith, 2020-11-28:
@houseofbugs the steps are nice, but without at least one of those configurations provided we don't know what the starting point is to begin executing those steps. Aside from disabling SEGMENT_LEVELED_MOVES there are a multitude of other options which we would be guessing at if not starting with one of your configs.

If you've tried it with so many different configurations we are just asking you to attach _one_ of them. It seems you feel like we are wasting your time, but you are asking someone else to volunteer to waste their own time guessing your configuration.

houseofbugs, 2020-11-28:
Similar report that is quite old: https://github.com/MarlinFirmware/Marlin/issues/14948

Whole firmware with configs: https://content.timothyhoogland.com/files/ShareXUpload/SegmentBug_V422_E3Board_2072Base.zip

This is my branch that's using 2.0.7.2. I deleted the bugfix test I had as I was just seeing if it was present on there or not as well, it was.

sjasonsmith, 2020-12-03:
I've reproduced this in the simulator. It is SegFaulting in a call to `mesh_bed_leveling::cell_index_x`, but I haven't investigated yet.

Steps to reproduce:
1. Use these configs and eeprom.dat in the simulator. It already has a mesh saved.
    [Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/5634379/Configuration_adv.zip)
2. `G28`
3. `M420 S1`
4. `G0 X1`

sjasonsmith, 2020-12-03:
This should be fixed in the above PR. If you are able to verify the change solves it on your system that would be great.

sjasonsmith, 2020-12-04:
Fix is merged. Please report back if there are still issues.

PR #20363 (2020-12-04): Fix MESH_BED_LEVELING w/o SEGMENT_LEVELED_MOVES
### Description

MESH_BED_LEVELING was hanging/crashing controllers when SEGMENT_LEVELED_MOVES is disabled, due to infinite recursion in `mesh_bed_leveling::line_to_destination`. This resolves the infinite recursion, and makes the MBL version of this function nearly (but not quite) identical to the ABL version.

### Benefits

Allows SEGMENT_LEVELED_MOVES to be disabled with MBL.

### Configurations

These are for use in the simulator. Any MBL setup could reproduce the problems by disabling SEGMENT_LEVELED_MOVES.
[Configuration_adv.zip](https://gi [... cut]
closed 2020-12-04 by sjasonsmith