MarlinFirmware/Marlin #20277: [BUG] M204 T5000 causes overacceleration

espr14, 2020-11-25:
### Bug Description

Too high travel speed causes in some moves extreme acceleration and speed. Seem like double or tripple of the max of the motors.

### Configuration Files

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5598767/Marlin.zip)

### Steps to Reproduce

1. Set some normal jerk (min.speed) / max. speed / acceleration for X motor.
1. Send M204 T9999
1. Run this G code
[test X.zip](https://github.com/MarlinFirmware/Marlin/files/5598770/test.X.zip)

**Expected behavior:**

Head moves smoothly and every move has the same acceleration. The longer moves have higher max. speed. No weird sounds.

**Actual behavior:**

Move to X70 and back to X10 is with extreme acceleration (much more than max. acc. of the motor) and it's the fastest speed of all moves of this test. It's sound like it's pushing its limits.

#### Additional Information

The same behavior is in bugfix-2.0.x. I had to turn off Junction deviation and S-curve because these were loosing steps much more than with classic jerk. However, I haven't tested this test with them. The same issue could be with M204 R and P.

sjasonsmith, 2020-11-28:
Is 9999 reasonable acceleration for your machine? If you are configuring your machine for acceleration higher than it can handle, then the behavior seems expected.

qwewer0, 2020-11-28:
Tested the test X.gcode, from the main comment, on the latest bugfix-2.0.x.

I have `LIMITED_MAX_ACCEL_EDITING` enabled and `MAX_ACCEL_EDIT_VALUES       { 1250, 1250, 100, 2000 }` set, so if I'm not wrong, the test after `M204 T9999` should behave the same as the test after `M204 T1250` (`DEFAULT_TRAVEL_ACCELERATION   1250`), but that doesn't seems to be the case. The test after the `M204 T9999` shows that the printer moves more violently than with `M204 T1250`.

I think the problem is that, marlin lest us change the max acceleration despite the max acc. edit limit.
I might misunderstood the issues main point, but this is how I interpreted it.

espr14, 2020-11-28:
[Video: https://youtu.be/vYh1qhB78uw](https://youtu.be/vYh1qhB78uw)

Reasonable acceleration: motors have it's own reasonable acceleration. The travel/print acceleration should be limited by the motor accelerations. What to set to ignore travel acceleration and follow motor acceleration only? M204 T0?

If the X motor has acceleration 1000 and the travel acc. is set to 9999, it should accelerate all the time by 1000. In reality, it does most of the time but sometimes it goes crazy so it goes over all limitations (speed, acc.). Similar happened with Z: max. speed 1200, G1 Z5 F9999 caused lost steps.

FW engine should compute next movement, look for limitations and then decrease speed (acceleration, jerk etc.) according to these. Ideally, it should double check motor's limitations if they can handle it. This would avoid numerical issue.

qwewer0, 2020-11-28:
I think my experience was the same as what you have in the video, so it doesn't moves with the same max acceleration all the time, as a result of the M204 T9999.

espr14, 2020-11-30:
I had to slow down to `M204 T1000` !!!
Similar issue with `G1 Z10 F9999` which is not about acceleration but speed.

qwewer0, 2020-11-30:
I modified @espr14's test file a bit: [max-acc_test.zip](https://github.com/MarlinFirmware/Marlin/files/5616141/max-acc_test.zip)

1. Simplified the file a bit
2. Shifted the X points so that it isn't too close to the endstop
3. Split it into two files (`YES-M204_max-acc_test.gcode`, `NO-M204_max-acc_test.gcode`)
4. Added `M502` to the beginning of both files
5. Added `M204 T9999` to `YES-M204_max-acc_test.gcode`, so that we don't need to send it to the printer manually

My configurations:
- `DEFAULT_MAX_FEEDRATE          { 150, 150, 10, 100 }`
- `MAX_FEEDRATE_EDIT_VALUES    { 150, 150, 10, 100 }` (**enabled** `LIMITED_MAX_FR_EDITING`)
- `DEFAULT_MAX_ACCELERATION      { 1000, 1000, 100, 2000 }`
- `MAX_ACCEL_EDIT_VALUES       { 1000, 1000, 100, 2000 }` (**enabled** `LIMITED_MAX_ACCEL_EDITING`)
- `DEFAULT_TRAVEL_ACCELERATION   1000`

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5616202/Configuration.zip)

Here are my results in two video: https://imgur.com/gallery/Wnn3dt5
(Bottom video has `M204 T9999` in it.)
By my understanding `M204 T9999` shouldn't be able to change the travel acc. higher than 1000, and both files should behave the same, but the file with `M204 T9999` behaves differently than the one without it. The file with `M204 T9999`, at around 40 sec in the video makes harsh movements (with higher acceleration?) compared to the other.

rhapsodyv, 2020-11-30:
Can anyone tell me a real situation where this bug is happen?

I know that it must be fixed. But reading the description seems more like a configuration error than a Marlin bug, because: why anyone would set max acc to a value, and then set the slicer to use a higher value?

qwewer0, 2020-11-30:
> Can anyone tell me a real situation where this bug happen?

Maybe someone using someone else's gcode from a different machine and settings, or without our knowledge the slicer inserts it into the gcode, like how PrusaSlicer did this in marlin gcode flavor, fortunately this got fixed (I think).

For me, I can't think any way that I would use a gcode that has that in it, but @espr14 might think differently.

> But reading the description seems more like a configuration error than a Marlin bug

Could you explain it in a bit more detail, to why do you think this is more like a configuration error than a Marlin bug? (don't need to be here, could be in discord)

espr14, 2020-11-30:
1. X motor can handle 4000 mm/s^2, Y motor 2000 mm/s^2. But I had to limit M204 to 1000 mm/s^2. Otherwise, it randomly overaccelerated and lost steps. Moreover, it lost steps on X axis!!!
1. M204 sets acceleration for all directions. If I set 2000, it won't use full power of the X motor. I would like to set 4000 and the engine should limit print/move acceleration to what motors handle.
1. How I tell: move as fast as possible? G1 F-1?

espr14, 2020-12-31:
Any updates on this? Could I enlarge the planner buffer for the algorithm to see more in the future?

espr14, 2021-03-05:
I was thinking of increasing buffer size but it should work with the smaller buffer.

nik-80, 2021-04-17:
I have same problem, I did notice that it is not because of extreme values, but at certain values
also same problem is noticed with or without S curve enabled
how behave on my printer - at certain values of M204 Z motor stall completely, even with very low acceleration set for Z
with M204 T1000 cant complete homing, on back off from first bump - it stall
with T900 and T1100 - homing is completed without stalling, but if I have next line with G0 Z10 (or any other value) - it stalls again
randomly stall after G4 instead of G28, but on G28 seems to be easy to reproduce
with T1300 - work fine, with 1400 fail again (same as 1000) 
with T1500 again work poorly, only complete homing, again fail randomly on G0
when I set very slow acceleration for Z around 10 and T1000 I can visually see that it is accelerating slowly when moving with manual controls, but while homing, the back off movement is executed with very high acceleration
in short something is messed up with M204, at certain values of T dont respect acceleration set for Z

nik-80, 2021-04-17:
more Im trying more weird it become
on manual control Im pressing home button and then move Z 10mm
with M201 Z200 and M204 T1200:
-home and while homing send move Z - work properly
-home and wait until finish, then move Z - it stalls
with M201 Z200 and M204 T1300:
- home, wait, move Z, wait, move Z again - it work
- home, dont wait, move Z, wait - until here it work fine, but if I move Z once again - it stalls

the above examples repeat consistently
very strange behaviour, which make me think the problem could be somewhere else, something overflow or overwrite existing data in memory or dont know, but very strange behaviour

espr14, 2021-06-17:
Let's keep it open.

espr14, 2021-08-18:
Let's keep it open for the last time.

sjasonsmith, 2021-12-20:
This might be related to other acceleration issues we’re looking at.

thisiskeithb, 2024-06-08:
Fixed in #27013

closed 2024-06-08 by thisiskeithb