MarlinFirmware/Marlin #20278: [BUG] Nozzle to rpobe offset ignored

daleckystepan, 2020-11-25:
### Bug Description

Nozzle to probe offset is ignored. Probably on X and Y axis.

### Configuration Files

Not relevant

### Steps to Reproduce

Compile latest 04c4c6004e752582235357b14aebdd5ff579f477 bugfix with bed leveling support.

**Expected behavior:**

Nozzle to probe offset will be correctly applied

**Actual behavior:**

Nozzle to probe offset is ignored.

#### Additional Information
Reverting 296a2ad7e45d88a10db4b3305ea183e5c0c177a5 this commit will fix the issue. https://github.com/MarlinFirmware/Marlin/commit/296a2ad7e45d88a10db4b3305ea183e5c0c177a5#r44537759 

Probe::offset is copied and then referenced instead of just referenced.

lococnc, 2020-11-25:
Confirmed, mine is ignoring it as well now.

thisiskeithb, 2020-11-25:
When reporting an issue, please be sure to follow all instructions at the top of the template, including debugging steps, and leave the sections headings intact.

For reference, these are the instructions included in the template:

### Before Reporting a Bug

1. Test with the `bugfix-2.0.x` branch to see whether the issue still exists.

2. Get troubleshooting help from the Marlin community to confirm it's a bug and not just a configuration error. Links at https://github.com/MarlinFirmware/Marlin/issues/new/choose

### Instructions

1. Fill out every section of the template below.

2. **Always attach configuration files, regardless of whether you think they are involved.**

3. Read and understand Marlin's Code of Conduct. By filing an Issue, you are expected to comply with it, including treating everyone with respect: https://github.com/MarlinFirmware/Marlin/blob/master/.github/code_of_conduct.md

qwewer0, 2020-11-26:
I have this problem too, on the latest bugfix-2.0.x

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5600746/Configuration.zip)

daleckystepan, 2020-11-26:
Sorry for that. Missing configuration files. 

https://github.com/daleckystepan/Marlin/blob/geeetech_a10/Marlin/Configuration.h

https://github.com/daleckystepan/Marlin/blob/geeetech_a10/Marlin/Configuration_adv.h

amertner, 2020-11-26:
I have this too, and the change @daleckystepan points to is definitely in the right area. Without being able to G34 and G29, my printer is unusable atm.

[Config.zip](https://github.com/MarlinFirmware/Marlin/files/5602476/Config.zip)

MarcoZ76, 2020-11-26:
i have the same problem...

Yannik25, 2020-11-26:
I have the same issue,
When undoing changes from commit 296a2ad it is working again
https://github.com/MarlinFirmware/Marlin/commit/296a2ad7e45d88a10db4b3305ea183e5c0c177a5

paulpenney, 2020-11-26:
> I have the same issue,
> When undoing changes from commit [296a2ad](https://github.com/MarlinFirmware/Marlin/commit/296a2ad7e45d88a10db4b3305ea183e5c0c177a5) it is working again

I can confirm that this worked for me as well to solve the issues.

bellini13, 2020-11-26:
I'm having the same issue.  If i use firmware from 11/24, it does work properly.  So I'm guessing a recent change has broken it

discip, 2020-11-26:
Same here:
<details>
  <summary>
    <b>Configuration.h</b>
  </summary>

```c#
/**
 * Marlin 3D Printer Firmware
 * Copyright (c) 2020 MarlinFirmware [https://github.com/MarlinFirmware/Marlin]
 *
 * Based on Sprinter and grbl.
 * Copyright (c) 2011 Camiel Gubbels / Erik van der Zalm
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program.  If not, see <https://www.gnu.org/licenses/>.
 *
 */
#pragma once

/**
 * Configuration.h
 *
 * Basic settings such as:
 *
 * - Type of electronics
 * - Type of temperature sensor
 * - Printer geometry
 * - Endstop configuration
 * - LCD controller
 * - Extra features
 *
 * Advanced settings can be found in Configuration_adv.h
 */
#define CONFIGURATION_H_VERSION 020008

//===========================================================================
//============================= Getting Started =============================
//===========================================================================

/**
 * Here are some standard links for getting your machine calibrated:
 *
 * https://reprap.org/wiki/Calibration
 * https://youtu.be/wAL9d7FgInk
 * http://calculator.josefprusa.cz
 * https://reprap.org/wiki/Triffid_Hunter%27s_Calibration_Guide
 * https://www.thingiverse.com/thing:5573
 * https://sites.google.com/site/repraplogphase/calibration-of-your-reprap
 * https://www.thingiverse.com/thing:298812
 */

//===========================================================================
//============================= DELTA Printer ===============================
//===========================================================================
// For a Delta printer start with one of the configuration files in the
// config/examples/delta directory and customize for your machine.
//

//===========================================================================
[... 4843 lines cut ...]
    #define E0_CHAIN_POS     -1
    //#define E0_INTERPOLATE true
  #endif

  #if AXIS_IS_TMC(E1)
    #define E1_CURRENT      800
    #define E1_MICROSTEPS    16
    #define E1_RSENSE         0.11
    #define E1_CHAIN_POS     -1
    //#define E1_INTERPOLATE true
  #endif

  #if AXIS_IS_TMC(E2)
    #define E2_CURRENT      800
    #define E2_MICROSTEPS    16
    #define E2_RSENSE         0.11
    #define E2_CHAIN_POS     -1
    //#define E2_INTERPOLATE true
  #endif

  #if AXIS_IS_TMC(E3)
    #define E3_CURRENT      800
    #define E3_MICROSTEPS    16
    #define E3_RSENSE         0.11
    #def

daleckystepan, 2020-11-26:
If you are experiencing same issue, just revert 296a2ad as mentioned in the first post.

sjasonsmith, 2020-11-26:
If one of you has located the cause of the problem please post a pull request to fix it.

tpruvot, 2020-11-26:
not exactly related, but the new LEVEL_CORNERS_USE_PROBE ignores it also
had to do that
```
-  #define LEVEL_CORNERS_INSET_LFRB { 30, 30, 30, 30 } // (mm) Left, Front, Right, Back insets
+  #define LEVEL_CORNERS_INSET_LFRB { 40, 30, 20, 30 } // (mm) Left, Front, Right, Back insets
```

sjasonsmith, 2020-11-26:
@daleckystepan thanks for your help troubleshooting this. I've merged the simplified fix in your PR, which seems to fix it. I'll leave this open until you have a chance to test it and verify it is working from the `bugfix-2.0.x` branch.

bill-orange, 2020-11-26:
Corrected my issue. Fast work.  Thanks.

sjasonsmith, 2020-11-27:
Closing, based on feedback from a couple people that it is now working. Please let us know if there are still issues related to this.

closed 2020-11-27 by sjasonsmith

daleckystepan, 2020-11-27:
Fixed also for me. Thank you for quick solution.