MarlinFirmware/Marlin #20281: [FR] G35 - Specifiable End Position

qwewer0, 2020-11-26:
### Description

At the moment `ASSISTED_TRAMMING_WAIT_POSITION` is only used in `ASSISTED_TRAMMING_WIZARD`, but it would be useful to add the `ASSISTED_TRAMMING_WAIT_POSITION` or similarly specifiable position with `G35` too, for easier screw adjustment.
The homing at the end is unnecessary (and a waste of time) if at the end the nozzle moves to a wait/safe position.

### Feature Workflow

1. Start `G35`
2. `G35` probes the bed
3. No homing happens (to not waste time)
4. Nozzle moves to `ASSISTED_TRAMMING_WAIT_POSITION` or similarly specifiable position (for easier screw adjustment)
5. End of `G35`

#### Additional Information

* Provide pictures or links that demonstrate a similar feature or concept.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

shitcreek, 2020-11-27:
Looking at `menu_tramming.cpp`, we have this bit:
```c++

  #ifdef ASSISTED_TRAMMING_WAIT_POSITION
    // Move XY to safe position
    if (DEBUGGING(LEVELING)) DEBUG_ECHOLNPGM("Moving away");
    const xyz_pos_t wait_pos = ASSISTED_TRAMMING_WAIT_POSITION;
    do_blocking_move_to(wait_pos, XY_PROBE_FEEDRATE_MM_S);
  #endif
```
It shouldn't be too hard to insert that into `G35.cpp`. I'm guessing it should be plugged into the ` LOOP_L_N` function like so:
```c++
  // Probe all positions
  LOOP_L_N(i, G35_PROBE_COUNT) {

    // In BLTOUCH HS mode, the probe travels in a deployed state.
    // Users of G35 might have a badly misaligned bed, so raise Z by the
    // length of the deployed pin (BLTOUCH stroke < 7mm)
    do_blocking_move_to_z((Z_CLEARANCE_BETWEEN_PROBES) + TERN0(BLTOUCH_HS_MODE, 7));
    const float z_probed_height = probe.probe_at_point(screws_tilt_adjust_pos[i], PROBE_PT_RAISE, 0, true);

    #ifdef ASSISTED_TRAMMING_WAIT_POSITION
      // Move XY to safe position
        if (DEBUGGING(LEVELING)) DEBUG_ECHOLNPGM("Moving away");
        const xyz_pos_t wait_pos = ASSISTED_TRAMMING_WAIT_POSITION;
        do_blocking_move_to(wait_pos, XY_PROBE_FEEDRATE_MM_S);
    #endif
```

shitcreek, 2020-11-27:
Perhaps we should also have a parameter to utilize the position or not:
```c++
if(parser.seen('P') do_blocking_move_to(wait_pos, XY_PROBE_FEEDRATE_MM_S);
```

qwewer0, 2020-11-27:
@shitcreek Thanks for the feedback.

If it is added to `LOOP_L_N`, wouldn't it move to `wait_pos` after each point is measured? (without a "last point" check)

> Perhaps we should also have a parameter to utilize the position or not:
> 
> ```c++
> if(parser.seen('P')) do_blocking_move_to(wait_pos, XY_PROBE_FEEDRATE_MM_S);
> ```

If there would be already and option to move away, why wouldn't someone want to use it? So I think it should move away without any extra parameter.

qwewer0, 2020-11-27:
Tested https://github.com/MarlinFirmware/Marlin/pull/20298 and it works, but I'm not sure if it needs some more work.

closed 2021-01-05 by qwewer0