MarlinFirmware/Marlin #20282: [BUG] Enabled DIRECT_STEPPING -> Blank screen

dj-fiorex, 2020-11-26:
### Bug Description

I'm on bugfix-2.0.x , i tried to build and install marlin after activating DIRECT_STEPPING, but once loaded on my printer the display remains blank.
I have a Eryone Thinker SE with MEGA2560, TMC2209, REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

### Configuration Files

https://we.tl/t-YzYHFsomQl

### Steps to Reproduce

1. Build Marling with DIRECT_STEPPING
2. Load on the printer
3. Turn on printer -> Blank screen

**Expected behavior:**

Screen is working normally

**Actual behavior:**

Screen is blank

ellensp, 2020-11-26:
I have attached the Config files here so you don't have to go to a weird site to get them.
[Configurations.zip](https://github.com/MarlinFirmware/Marlin/files/5603373/Configurations.zip)

dj-fiorex, 2020-11-26:
Thanks @ellensp, this is my first issue here!
Looking forward to get a solution, because i have a spare pi4 and want to try direct stepping

ellensp, 2020-11-26:
I compiled your configs and the first thing I noticed is, this configs exceeds the available ram of the controller.
RAM:   [==========]  **117.2%** (used 9600 bytes from 8192 bytes)
Flash: [=======   ]  70.6% (used 179216 bytes from 253952 bytes)
You must reduce the the amount of used ram to less than 100% by disabling some features.

dj-fiorex, 2020-11-26:
Oh, you are right!
Ram before enable DIRECT_STEPPING: 84% (6878 bytes)
Ram after enable DIRECT_STEPPING: 134.7% (11038 bytes)
What can i disable to try DIRECT_STEPPING?

daleckystepan, 2020-11-26:
You can try disable SD write, G2/G3 ARC, TMC_DEBUG and other debugs.

dj-fiorex, 2020-11-26:
Nothing to do, tried disabled SDSUPPORT, ARC_SUPPORT, TMC_DEBUG, applied 3x3 grid instead of 5x5,  always out of ram :(

Bastlwastl84, 2020-11-26:
Perhaps this guide helps you https://crosslink.io/2020/08/14/shrinking-marlin-2-0-how-to-reduce-firmware-size-for-8-bit-boards-and-still-use-a-bltouch-and-the-filament-sensor/

ellensp, 2020-11-26:
Your controller has 8k of ram. From the before and after enabling DIRECT_STEPPING it can be determined that DIRECT_STEPPING uses 4160 bytes of ram, ie over 4k.
You would have to turn off practically everything.

Starting with a defatult config provided with marlin and just enabling the LCD and direct_stepping 
RAM:   [========= ]  89.1% (used 7296 bytes from 8192 bytes)

closed 2020-11-26 by ellensp