MarlinFirmware/Marlin #20285: [BUG] ABL disregards probe offsets

sdaitzman, 2020-11-26:
### Bug Description

ABL attempts to probe at the expected positions with the nozzle, rather than the BLTouch. In other words, when leveling using the BLTouch, the printer moves the nozzle to the center then probes there (rather than moving the nozzle OFF CENTER, with the BLTouch over center).

I'm running on an Ender 3 Pro with an SKR E3 DIP. My Marlin lives at https://github.com/sdaitzman/marlin

### Configuration Files

**Required:** Include a ZIP file containing `Configuration.h` and `Configuration_adv.h`.

[configs.zip](https://github.com/MarlinFirmware/Marlin/files/5604106/configs.zip)

If you've made any other modifications describe them in detail here.

### Steps to Reproduce

1. Compile firmware and upload it with these modifications (I just merged from bugfix-2.0.0 to confirm).
2. Menu navigate to "bed level" and confirm
3. Watch as printer homes Z in the wrong place on the bed, then crashes its nozzle repeatedly into the bed while the BLTouch hangs off the side.

**Expected behavior:**

Bed leveling places the probe in expected positions, not the nozzle.

**Actual behavior:**

Bed leveling places the nozzle where the probe should be.

#### Additional Information

Can provide a video if that's helpful. Thanks!

daleckystepan, 2020-11-26:
Duplicate of 20278

tpruvot, 2020-11-26:
not only ABL... UBL too

thisiskeithb, 2020-11-26:
Duplicate of #20278

closed 2020-11-26 by thisiskeithb