MarlinFirmware/Marlin #20287: [BUG] Blobs and stuttering Anet et4, MCU: STM32F407VGT6 ARM CORTEX M4

Daniel-br0, 2020-11-26:
[image]

### Bug Description

### Configuration Files

**Required:** Include a ZIP file containing `Configuration.h` and `Configuration_adv.h`.

If you've made any other modifications describe them in detail here.

### Steps to Reproduce

1. Compiling Marling 2.0 bugfix for Anet et4+ with inductive sensor using platformio, https://github.com/davidtgbe/Marlin
2. Flashing bootloader to 0800000 and than .srec to 08010000
3. Printing objects like julians vase with complex mesh https://www.thingiverse.com/thing:126567.

**Expected behavior:**

Expect nice and smooth surface.

Instead I have blobs on surface.It is not a bad filament with humidity or cura config with min mesh set to 0,5 (cnc kitchen), tried with prusa slicer, simplify and cura, the same. Printing from a good sd card, ajusted buffers to 32 instead of 16, almost no changes. My guess -it has something to do with CPU, but I don't know where to look for exactly.

[image]

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [Contributing to Marlin](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md) for additional guidelines.

shitcreek, 2020-11-26:
You are using a separate branch of Marlin. Try the official bugfix and see if you get the same results.

closed 2020-11-26 by Daniel-br0

thisiskeithb, 2020-11-27:
The ET4 motherboard is not supported here yet, but there’s a pending PR to add support: https://github.com/MarlinFirmware/Marlin/pull/20280