MarlinFirmware/Marlin #20288: MKS TS35 calibration each time

srastoul, 2020-11-26:
**Bug Description :**

Since the download of the last version of Marlin bugfix 2.0.x each time I start the screen asks me for a calibration. The version for the month of october only requires a calibration at the first start

**Configuration Files:**

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5604450/Marlin.zip)

**Steps to Reproduce:**

Shutdown or reset 3d printer

**Additional Information:**

Bord MKS SGEN-L V2.0

[image]

altiber, 2020-11-26:
I had the same problem. After calibration, you need to save the settings via the menu in EEPROM. This should help

rhapsodyv, 2020-11-26:
If you have `EEPROM_SETTINGS` disabled, you need to connect to serial, run touch calibration, that the numbers it will show, and put that in your config.

If you have `EEPROM_SETTINGS` enabled, you must run `M500` after touch calibration, to save it on EEPROM.

Your configs shows that you don't have `EEPROM_SETTINGS` enabled, so you need to copy the values shown in the serial to your `Configuration.h`

closed 2020-11-26 by rhapsodyv