MarlinFirmware/Marlin #20289: [BUG] BigTreeTech SKR 1.4 Turbo + generic SSD1306 I²C 1.3" OLED : shifted display

yet-another-average-joe, 2020-11-26:
### Bug Description

Brand new BTT MKS 1.4 Turbo with :
- 5x BTT TMC2209 drivers
- 2x generic NTCs
- a generic 1.3" I2C OLED (4-wire version)
- a generic click encoder
- a 12V brick for testing purpose
- vanila Marlin Bugfix (62680bb) (downloaded 2 days ago)

Test on the bench : no motors, just the drivers so marlin does not complain.

I2C pullups are on the board, and probably on the OLED (difficult to say without a microscope or the schematics, and for obvious reasons I don't want to test with an ohm-meter)

### Configuration Files

Display defineded as "OLED_PANEL_TINYBOY2" (for the encoder) 

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5604497/Marlin.zip)

### Steps to Reproduce

Turn it ON, then wait from a few minutes to 15 minutes.

**Expected behavior:**

Informations being displayed properly.

**Actual behavior:**

After a few minutes to 15 minutes, the text and graphics are shifted down by 8 pixels, and the bottom is displayed on top. 

I also experimented some crashes (reboots), but it is not reproductible. (and while I'm writing this, I got the first one after 1H05 !

Playing with menus makes no changes : display remains shifted. (in fact, it appears rolled)

[image]

#### Additional Information

As is, such a setup is not usable.

It seems to be a known issue, but very few are using such displays...
I also have similar issues with a generic 0.96" SSD1306 OLED (also shifted to the right). Found a fix for the horizontal shift, but not for the vertical one (tinkering with u8glib). Crashes (reboots) every 5mn or so.
These displays are good : played a lot with them, but using u8g2. It seems they are supported with u8glib2, not really with u8glib, and chances are they will never be as Marlin relies on u8glib, and u8glib support ended years ago.
Soon, I will test a third display : generic 1.54" SSD1306 OLED.

BTW, if there is no solution to this problem, are there some known good SPI OLED displays ? Are I²C OLEDs doomed ?

rhapsodyv, 2020-11-27:
I will close this, as it seems to be duplicated of #14431.

Thanks!

closed 2020-11-27 by rhapsodyv

sjasonsmith, 2020-11-28:
Duplicate of #14431