MarlinFirmware/Marlin #20295: MKS_SGEN_L Expansion pin for additional steppers

theomajigga, 2020-11-27:
Unable to use the following pins for external STEP/DIR pins, using Latest release:

P1.28
P1.26
P1.24
P0.25
P0.02 (TX)
P0.03 (RX)

Pins P1.23 and P2.0 work (which also means the stepper extensions board is working).

Trying to add 2 more steppers, (Dual Z and triple extruder)

ellensp, 2020-11-27:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-11-27 by ellensp

theomajigga, 2020-11-27:
This isn't a configuration issue, or a troubleshooting issue, I have tested wires, multiple pins (hence the tag that some are working)

ellensp, 2020-11-27:
Yes many say that, then they enable pins_debugging and find the conflicts with things they didn't expect to be on those pins. remove them and suddenly it works. 
You also didn't follow the template or attach config files. so that makes this a support request.

ellensp, 2020-11-27:
But in his case I suspect its purely hardware issues.

P1.28 and P1.26 are hard wired to diag pins., if you have a stepper driver in E0 or E1 with this pin, you must remove the diag pin jumpers to disconnect them.
P1.24 and P0.25 both have additional circuity on the controller. You would have to remove components from the board to use these pins.

"Pins P1.23 and P2.0 work" yes these don't have additional circuitry on these pins.