MarlinFirmware/Marlin #20297: [FR] Multiple Serial Port Host Support for AVR chips

Jovian-Dsouza, 2020-11-27:
Many AVR chips have more than one hardware serial support (Ramps 1.4 in my case) and I would like to attach a secondary host (such as the ESP3D) along with the USB to Serial host . I tried using the SERIAL_PORT_2 as my secondary port , but it does not seem to work. Is the `SERIAL_PORT_2` supported in Marlin ? 

```
SERIAL_PORT  0 
SERIAL_PORT_2  2
BAUDRATE 250000
```

ellensp, 2020-11-27:
There are 4 serial ports on a mega2560, on the ramps they are mapped to the following.

2 RXD0 PE0 ( RXD0/PCINT8 ) Digital pin 0 (RX0) usb
3 TXD0 PE1 ( TXD0 ) Digital pin 1 (TX0) usb
45 RXD1 PD2 ( RXDI/INT2 ) Digital pin 19 (RX1) Z-MAX
46 TXD1 PD3 ( TXD1/INT3 ) Digital pin 18 (TX1) Z-MIN
12 RXD2 PH0 ( RXD2 ) Digital pin 17 (RX2) on aux4
13 TXD2 PH1 ( TXD2 ) Digital pin 16 (TX2) on aux4
63 RXD3 PJ0 ( RXD3/PCINT9 ) Digital pin 15 (RX3) Y-MAX
64 TXD3 PJ1 ( TXD3/PCINT10 ) Digital pin 14 (TX3) Y-MIN

And yes they all work.

Ramps is 5v, ESP3D is 3.3v, first time you send data from mega to the ESP3D it will burn out the ESP3D RX port.

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-11-27 by ellensp

Jovian-Dsouza, 2020-11-27:
Sorry for the inconvience  and thanks for your comments , I had remap some endstop pins to make it work on Serial 1. Here is what worked for me

```
#define SERIAL_PORT 0

/**
 * Select a secondary serial port on the board to use for communication with the host.
 * :[-1, 0, 1, 2, 3, 4, 5, 6, 7]
 */
#define SERIAL_PORT_2 1

//NOTE : Connect Z endstop to Y_MAX  as Z
//dont use Z_MIN and Z_MAX for endstop
//instead connect Z_MIN (TX1) and Z_MAX(RX1)

#define Y_MAX_PIN          -1
#define Z_MIN_PIN          15
#define Z_MAX_PIN          -1
```