MarlinFirmware/Marlin #20304: [BUG] Probe (bltouch) not stowed before homing xy

hamster65, 2020-11-27:
**Expected behavior:**
Probe is stoved before x, y, or z is moved during homing.

**Actual behavior:**
Printer homes x and y and does not stow probe before. It stows before moving to the middle of the bed to home Z.

### Steps to Reproduce
- Deploy probe
- G28
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5608915/Marlin.zip)

sjasonsmith, 2020-11-27:
I don’t think this is a bug.

This is a case of “don’t do that”.

If you manually deploy the probe and then tell Marlin to move, that is what it does. It only alters commands to the BLTouch when you have actually instructed it to use the BLTouch.

hamster65, 2020-11-27:
Yes, that's one way to look at it. On the other hand, I think marlin should make sure nothing gets damaged during homing. That's why there are functions for lowering the bed before homing etc. That includes bringing the probe into a safe state.

I should have mentioned that I did not manually / intentionally deploy the probe before G28. It just happened to be deployed because of something I did before. I'll try to figure out what exactly that was. Probably levelling or tramming.

sjasonsmith, 2020-11-27:
If something that uses the probe fails to stow that probe, _that_ would be a bug.

sjasonsmith, 2020-11-28:
@hamster65 based on the information we have now, this does not appear to be a bug, so I'm closing it as such. Marlin simply doesn't really have probe state outside of the command actually using the probe. It is the responsibility of each command to ensure the probe is stowed upon completion or error.

If you are able to isolate actions which leave the probe deployed when they should not, please create a new report for that with specific instructions to reproduce.

closed 2020-11-28 by sjasonsmith