MarlinFirmware/Marlin #20305: [BUG] start was changed to echo:start

repetier, 2020-11-27:
### Bug Description

Standard says that firmware starts with
start

but when I used latest commit I saw it now says
echo:start

which is against standards and can cause hosts to not detect a reset of printer if they watch for a line beginning with "start" and not a line containing "start" which could also be a file in listing or a M118 response or just a word in a message. 

### Configuration Files

Happens for all configs

### Steps to Reproduce

Connect

**Expected behavior:**

Start communication with
start

**Actual behavior:**

Starts communication with
echo:start

#### Additional Information

Violation against many years old standard for communication.

ellensp, 2020-11-27:
See https://github.com/MarlinFirmware/Marlin/issues/19127

repetier, 2020-11-28:
Answer here since other issue is locked.
https://reprap.org/wiki/G-code#Replies_from_the_RepRap_machine_to_the_host_computer

> The most common response is simply:
> ok
> When the machine boots up it sends the string
> start

Hosts also show non echo commands. Actually many firmwares do not even use echo: at all. But "start" is a control command just like "ok". As you already saw in the referenced thread simplify3d has problems with it. Same with other hosts like our software. 

So what you are doing is changing a many years old standard and expect all existing hosts to register the change and update and all users using newer marlin version to update the host software to make it work correctly instead of removing the 5 bytes added in Marlin? Such a thing is called breaking backward compatibility and that should be done for a good reason!

rhapsodyv, 2020-12-03:
The fix was merged. Thanks!

closed 2020-12-03 by rhapsodyv