MarlinFirmware/Marlin #20306: [BUG] MMU2S C0 loop when loading

kakou-fr, 2020-11-27:
I am trying the last bugfix 2.0  code, if  i try to load a filament, and the filament cannot be engaged in the gear, the process never stop and the printer is blocked in an infinity loop

```
Recv: MMU => 'ok' (filament not present in gears)
Recv: MMU <= 'C0' (keep trying)
[...]
Recv: MMU => 'ok' (filament not present in gears)
Recv: MMU <= 'C0' (keep trying)
[...]
Recv: MMU => 'ok' (filament not present in gears)
Recv: MMU <= 'C0' (keep trying)
[...]
Recv: MMU => 'ok' (filament not present in gears)
Recv: MMU <= 'C0' (keep trying)
[...]
Recv: MMU => 'ok' (filament not present in gears)
Recv: MMU <= 'C0' (keep trying)
[...]
Recv: MMU => 'ok' (filament not present in gears)
Recv: MMU <= 'C0' (keep trying)
[...]
Recv: MMU => 'ok' (filament not present in gears)
Recv: MMU <= 'C0' (keep trying)
[...]
Recv: MMU => 'ok' (filament not present in gears)
Recv: MMU <= 'C0' (keep trying)
[...]
Recv: MMU => 'ok' (filament not present in gears)
Recv: MMU <= 'C0' (keep trying)
[...]
Recv: MMU => 'ok' (filament not present in gears)
Recv: MMU <= 'C0' (keep trying)
[...]
Recv: MMU => 'ok' (filament not present in gears)
Recv: MMU <= 'C0' (keep trying)
[...]
Recv: MMU => 'ok' (filament not present in gears)
Recv: MMU <= 'C0' (keep trying)
[...]
Recv: MMU => 'ok' (filament not present in gears)
Recv: MMU <= 'C0' (keep trying)
```

ellensp, 2020-11-27:
Please follow the Issue Template when filing a bug report. This includes attaching your configs:
 
```lisp
<!--

Have you read Marlin's Code of Conduct? By filing an Issue, you are expected to comply with it, including treating everyone with respect: https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/.github/code_of_conduct.md

Do you want to ask a question? Are you looking for support? Please don't post here. Instead please use the Marlin Firmware forum at http://forums.reprap.org/list.php?415 or the Marlin Facebook Group https://www.facebook.com/groups/1049718498464482/ or the Marlin Discord Server https://discord.gg/n5NJ59y.

Before filing an issue be sure to test the 1.1 and/or 2.0 "bugfix" branches to see whether the issue is already addressed.

-->

### Description

<!-- Description of the bug or requested feature -->

### Steps to Reproduce

<!-- If this is a Bug Report, please describe the steps needed to reproduce the issue -->

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.
```

kakou-fr, 2020-11-29:
i updated the post with the template and config file

closed 2021-01-06 by github-actions[bot]