MarlinFirmware/Marlin #20307: [BUG] ABL can not be activated in the latest bug fix fro Ender 3 v2

kyraxyg, 2020-11-27:
### Bug Description

In the latest bugfix, I have activated the ABL bed leveling in the configuration files. When I try to activate ABL in the G28-G29-M500 sequence, the printer gets restarted end gets a DWIN handshake error. I can not save a mesh. It seems the issue is also applicable to UBL case.
As seen from my research beforehand to diagnose the issue.
https://www.reddit.com/r/ender3v2/comments/jxuskb/bltouch_ubl_saving_mesh_crashes_the_printer/

Send: M500
Recv: echo:start
Recv: PowerUp
Recv: Marlin bugfix-2.0.x
Recv: 
Recv: echo: Last Updated: 2020-11-27 | Author: YcLn
Recv: echo:Compiled: Nov 27 2020
Recv: echo: Free Memory: 49079  PlannerBufferBytes: 1600
Recv: 
Recv: DWIN handshake error.
Recv: echo:EEPROM Initialized
Recv: BL24CXX Check succeeded!
Recv: echo:SD card ok

### Configuration Files

**Required:** Include a ZIP file containing `Configuration.h` and `Configuration_adv.h`.

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5609271/Marlin.zip)

If you've made any other modifications describe them in detail here.

### Steps to Reproduce

1. [First Step] Execute G28
2. [Second Step] Execute G29
3. [and so on...] Execute M500

**Expected behavior:**
Saving the mesh in eprom

**Actual behavior:**
Printer soft reboots and gives DWIN Handshake error. I can not determine whether the DWIN hasnshake is the cause or the result

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [Contributing to Marlin](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md) for additional guidelines.

kyraxyg, 2020-11-28:
[serial.log.txt](https://github.com/MarlinFirmware/Marlin/files/5610483/serial.log.txt)
.Here is an serial log file from octo print. The printer soft reboots after M500. This is also the case if I level from the LCD screen.

sjasonsmith, 2020-11-30:
I believe this is happening because the watchdog timer is expiring while writing to the EEPROM. The EEPROM write is optimized to skip unchanged bytes, which is why you only see this when storing your large 10x10 mesh after updating it.

kyraxyg, 2020-11-30:
The thing is I was writing 15x15 UBL mesh with success previously. But I tried 10x10 UBL and UBL also seems to fail . Is there a way for me to increase watchdog timeout and try. I don't want to disable watchdog as this might be dangerous.

sjasonsmith, 2020-11-30:
I am working on a fix that will service the watchdog during long writes.

sjasonsmith, 2020-11-30:
@kyraxyg can you test the change from the pull request above to see if it fixes your issue?

kyraxyg, 2020-11-30:
I will check when I get back home from work. Thanks

kyraxyg, 2020-11-30:
I have tested with this commit and test was successful. I have sucessfully saved my 10x10 mesh to eprom.

kyraxyg, 2020-11-30:
Thanks for the support

closed 2020-12-01 by sjasonsmith