MarlinFirmware/Marlin #20309: [BUG] Change Media does nothing

qwewer0, 2020-11-28:
### Bug Description

After the PR https://github.com/MarlinFirmware/Marlin/pull/20176, there is no longer a "Release Media", instead there is a "Change Media" menu item, that doesn't appear to do anything.

SKR Mini E3 v1.2, **non USB** `STM32F103RC_btt` / `_512k` environment.

I opened this issue not to get back the "Release Media", but instead because there is now a menu item that does nothing.
If on a **non USB** environment (e.g. `STM32F103RC_btt_512K`) the SD card can be safely removed without releasing the media, then I think there is no reason to have "Release Media" or "Change Media" menu item (I prefer this option).
But if this isn't the case, then we should have either "Release Media" or "Change Media" menu item that is working.

### Configuration Files

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5610403/Configuration.zip)

### Steps to Reproduce

0. Flash the latest bugfix-2.0.x (after https://github.com/MarlinFirmware/Marlin/pull/20176)
1. Insert SD card
2. Select "Change Media" menu item
3. Observe that nothing happens

**Expected behavior:**

Not having either a "Release Media" or "Change Media" menu item.

**Actual behavior:**

There is a "Change Media" menu item, that does nothing.

rhapsodyv, 2020-11-28:
In theory, all we did was remove the section that did “undef SD_DETECT_PIN”. But seems it have a colateral effect. 

I will take a look.

qwewer0, 2020-11-28:
@rhapsodyv Yes, I remember, I tried to look at the changes, but couldn't find something that could affect this.
Thank you very much for the check.

rhapsodyv, 2020-11-28:
Well ... it's confusing ... but there's a reason ...

Point 1) Before that PR, all boards with `SD_DETECT` were being ignored. So, even if you have a `SD_DETECT_PIN`, it wasn't being used by Marlin.

Point 2) Marlin have different behavior when you have SD_DETECT and you when don't have it.

a) When you ** don't ** have `SD_DETECT`: you need manually release the media, because Marlin will not be able to detect when you removed it. So, here we have the `Release Media` menu, that just does a` M22`.

b) When you have `SD_DETECT`: Marlin will automatically detect when the SD card was pulled off. So, it automatically call its internal release routines. So Marlin don't need that you manually release it in the menus. Because of that, it change the menu to "Change Media", that just does a "M21" to re-init it. It even don't need to be called, just to use force a re-init if it gave any error on automatic marlin code.

The "Change Media" is a new thing to a lot of us, as it was almost always disabled because of the bug with `SD_DETECT` that was fixed by # 20176 ....

So, in fact it's not a bug. Maybe a confusing menu ... I think we can even remove it, or maybe rename to "Refresh Media", or "Restart Media" ...

But the behavior seems correct, although confusing ...

qwewer0, 2020-11-28:
Refresh Media sounds better, more telling.
While I would love to see and option to disable that menu item, but I did not thought it through to what disadvantages would the lack of it mean.

@rhapsodyv What do you think, close this and open a FR to rename Change Media to Refresh Media, or change this issue into the FR?

rhapsodyv, 2020-11-28:
Maybe removing it at all would be better. 
I think we already have a "Refresh Media" inside the file browser, or not?
I will take a look in all options. 

Good point. I will change this to FR.

qwewer0, 2020-11-28:
There is a "Refresh", but that is only that:
https://github.com/MarlinFirmware/Marlin/blob/e8177735a00d1d7bb20b439544d3324fe01cfe7e/Marlin/src/lcd/language/language_en.h#L346

If removing it wouldn't mean any negative effects and that is the easiest way to handle it, than so be it.

qwewer0, 2020-11-28:
@rhapsodyv I have no idea how to describe this as a FR, so I just quoted and linked two of your comment.

rhapsodyv, 2020-11-28:
It's ok this way. I think I will work on this soon.

mathiasvr, 2020-12-01:
This reminds me of #17272, where I also wanted to remove all these types of menu items but it wasn't so simple to support all types of configurations. Still hoping this can be simplified somehow. 🙂

billbo1958, 2020-12-22:
I would like to cast a vote for renaming the menu item 'Change Media' to 'Refresh Media', rather than removing the menu item entirely.  The 'Change Media' menu item may appear to do nothing, but I found out that it does actually refresh the media.  (Please excuse me if I am not as clear as I ought to be - this is my very first 'Github' posting and I am not a firmware programmer.)
 
 I installed the latest Marlin 2.0 with bug fixes to my Ender 5 Pro printer, because it looked to me like the firmware that Creality provides for this printer does not include all of the menu items that I am used to seeing in Marlin.  (Also, Creality does not provide the source code for the Ender 5 Pro firmware; you can only download different versions of compiled firmware from their website.)  The Ender 5 Pro comes with a micro SD card slot; I purchased a micro-SD to standard SD adapter so that I could use a full size SD card with the printer.  I successfully configured and installed the Marlin firmware, and I noticed that if I put an SD card into the adapter and then plugged the adapter into the micro SD slot, the display read 'Media Inserted', and I could then print any of the files on the SD card.  If I removed the SD card (and left the adapter plugged in), the display would still show 'Media Inserted', and it would still display the same list of files if I selected the menu item, 'Print from media'.  If I put more files on the SD card and re-inserted in into the adapter, I noticed that the list of filed displayed would not show the new files I added.  When I clicked on the 'Change Media' menu item, I found that the list of files would update and show the new files I added.  If the 'Change Media' menu item were eliminated, the only way to 'refresh' the media would be to either unplug the adapter and plug it in again, or turn the printer off and on again.  If anyone has any questions about this or wants to know more about how the media menu behaves when using an SD to micro SD adapter, let me know.

idonasch, 2020-12-22:
After reading through this issue and believing to have understood the behavior of this menu item, I propose using the past form: `Changed Media ` as it is already used in the German translation which makes more sense to me.
Just my 2ct

gaby64, 2021-03-04:
I would also like to be able to hide the Change Media menu item

thisiskeithb, 2021-09-24:
Duplicate of https://github.com/MarlinFirmware/Marlin/issues/16912

closed 2021-09-24 by thisiskeithb