MarlinFirmware/Marlin #20310: [BUG] Unable to set esteps above 186 on Ender 3 v2 screen

CRCinAU, 2020-11-28:
### Bug Description
When entering esteps via the screen on an Ender 3 v2, a value above 186 is not able to be entered.

This means that when using some popular geared extruders that require the e steps to be configured to ~421, this can only be done via g-code.

### Configuration Files
https://github.com/MarlinFirmware/Configurations/tree/import-2.0.x/config/examples/Creality/Ender-3%20V2

### Steps to Reproduce
Attempt to set an estep value to above 186 in the screen UI.

This has been known for a while, but I can't locate any bug report to get this issue addressed. The workaround is to set the esteps via gcode, and then save in the EEPROM, however this is kind of clunky.

sjasonsmith, 2020-11-29:
I don't have one of these screens up to test with, but here is something you can try:

Look for the following code in `src/lcd/dwin/e3v3/dwin.cpp`:
```
    // Step limit
    if (WITHIN(HMI_flag.step_axis, X_AXIS, E_AXIS))
      NOMORE(HMI_ValueStruct.Max_Step, default_axis_steps_per_unit[HMI_flag.step_axis] * 2 * MINUNITMULT);
```

and replace it with something like this:
```
    // Step limit
    if (WITHIN(HMI_flag.step_axis, X_AXIS, E_AXIS))
      NOMORE(HMI_ValueStruct.Max_Step, 9999.99 * MINUNITMULT);
```

It looks like it defaults to only allowing you to double the default steps/mm. I don't know how this display presents this adjustment, so maybe there is an advantage to a smaller range. Hopefully this will at least get you closer to coming up with a good solution for it.

sjasonsmith, 2020-11-29:
Unfortunately this won't provide the ability to invert the motor, as we discussed yesterday. That still has to be done through the Configuration.h or by moving wires.

sjasonsmith, 2020-11-29:
I happened to already have a 4.27 board set up, so I attached the display. I'll post a PR right now to get it up to 999.9. Additional layout work will be needed if there is a desire to go beyond that, since the widths of fields change.

sjasonsmith, 2020-11-30:
@CRCinAU I've merged that change and will go ahead and close this issue. Please let us know if any issues arise with that.

closed 2020-11-30 by sjasonsmith

CRCinAU, 2020-11-30:
I feel that as long as it can go above ~450, it'll be fine for every case I've seen to date. I'm sure someone can raise a bug if some exotic setup with this screen requires more than 999.9...

sjasonsmith, 2020-11-30:
If these screens end up widely available then I'm sure it will be needed. As it is you cannot even buy a replacement from Creality, they are available _only_ with a new printer. I got lucky enough to find on on eBay, but I doubt we will see very many of them not on an Ender 3.