MarlinFirmware/Marlin #20311: [BUG] M190 S<temp> trigger COLD_EXTRUSION

Minims, 2020-11-28:
### Bug Description

M190 S<temperature> trigger message "cold extrusion prevented" 
But this gcode is for BED Temp

### Configuration Files

[Config.zip](https://github.com/MarlinFirmware/Marlin/files/5610830/Config.zip)

### Steps to Reproduce

1. Start the printer
2. Run M190 S60
3. Printer message "echo: cold extrusion prevented
4. Printer is poweroff

**Expected behavior:**

Printer heat the bed at 60°
 
**Actual behavior:**

Printer print message "cold extrusion prevented" and then halted.

#### Additional Information

Running Core XY 3Z on SKR Pro 1.1

qwewer0, 2020-11-28:
I wasn't able to reproduce the issue. While I have a different board and printer I still wanted to try it.
SKR Mini E3 v1.2, Ender 3, latest bugfix-2.0.x
I received: 
```
//action:notification Ender-3 Ready.
//action:notification Bed Heating...
```
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5610874/Configuration.zip)

Minims, 2020-11-28:
Found, it was the pull-up activated instead of pull-down on power resume pin, but the message in not clear. why cold extrusion ?

closed 2020-11-28 by Minims