MarlinFirmware/Marlin #20312: Latest bugfix has no touchscreen function on Anycubic Predator

rq3, 2020-11-28:
### Bug Description
Using Platformio under Visual Studio, successfully compiled the latest bugfix using the import example configuration and configuration_adv files from the delta/anycubic/predator directory.
The created bin file loads to the printer, but the printer does not respond to touch input at all, and the icon colors and orientation are incorrect.

### Configuration Files
[Config_Files.zip](https://github.com/MarlinFirmware/Marlin/files/5610900/Config_Files.zip)

### Steps to Reproduce

1. Download latest bugfix
2. Download latest import configurations from https://github.com/MarlinFirmware/Configurations/tree/import-2.0.x/config/examples/delta/Anycubic/Predator
3. Modify platformio.ini to reflect correct trigorilla_pro board
4. Successfully compile 
5. Successfully load created bin file to Anycubic Predator

**Expected behavior:**
To have the Predator touchscreen function when touched, and display the correct icons

**Actual behavior:**
There is no touchscreen response, and the icons are incorrect (mirrored, wrong colors)

#### Additional Information
[Predator_Screen](https://user-images.githubusercontent.com/3343927/100520663-e699cc00-316c-11eb-8dd6-6f0db4a2f0a9.JPG)

rhapsodyv, 2020-11-28:
Just run M995 to calibrate the touch. 

I think those values from config are outdated.

rhapsodyv, 2020-11-28:
Can you do this?

1. Connect to serial from pronterface or octoprint, run M995, calibrate your touch, and send the values here?

2. Test TFT_COLOR_UI and send a picture?

Thanks

rq3, 2020-11-28:
TFT_COLOR_UI doesn't compile, so I couldn't test that. Running M995 from Repetier Host yielded:

#define TOUCH_CALIBRATION_X  16810
#define TOUCH_CALIBRATION_Y -11570
#define TOUCH_OFFSET_X                 -13
#define TOUCH_OFFSET_Y                 338

which brought the touch functions to life. Just minor issues with icon colors and orientation as seen in my original post photo.

Just a thought: there is very little info on how to invoke M995 and what the results mean or what to do with them. Even comments in configuration.h would be of huge benefit. I believe I know that the X and Y offsets represent pixel displacement from an origin (0,0), but the meaning of the x and y calibration factors is still opaque to me. 

I know there has been a huge effort lately to "unify" how Marlin deals with the quadrillions of different displays available, so progress is being made!

sjasonsmith, 2020-11-28:
>  Just minor issues with icon colors and orientation as seen in my original post photo.

What was wrong here? I couldn't see anything wrong in the picture, but maybe just don't know what I am looking for?

rhapsodyv, 2020-11-28:
> > Just minor issues with icon colors and orientation as seen in my original post photo.
> 
> What was wrong here? I couldn't see anything wrong in the picture, but maybe just don't know what I am looking for?

@rq3 
That is the reason I asked for Color UI. As it's colourful, it's easy to check color issues.

> #define TOUCH_CALIBRATION_X 16810
#define TOUCH_CALIBRATION_Y -11570
#define TOUCH_OFFSET_X -13
#define TOUCH_OFFSET_Y 338

Can you open a PR on Configurations repository, changing the old calibrations values to the new ones you found?

rq3, 2020-11-28:
I wish I could have run TFT_COLOR_UI, but it wouldn't compile, with the error I e-mailed you.

The current bugfix screen is in the photo in my original post. The selection buttons are red, and the down arrow increments, while the up arrow decrements, which is a tad confusing.

This is what Marlin 2.0.7.2 looks like on my printer:
[image]

rq3, 2020-11-28:
@rhapsodyv  wrote: Can you open a PR on Configurations repository, changing the old calibrations values to the new ones you found?

I've been waiting until the whole display issue thing stabilizes in Marlin before diving in to a PR. The existing configuration files for the Anycubic Predator delta are based on an initial release that has nothing to do with the stock machine. Those files cannot function on a stock Predator with the Trigorilla Pro board, and I'd rather do a PR on a fully functional pair of configuration files, as opposed to multiple PR's while chasing a moving target.

Here's the config files that work on my stock machine under Marlin 2.0.7.2:
[2072_Stock_Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5611226/2072_Stock_Configuration.zip)

rhapsodyv, 2020-11-28:
Ok, touch is "solved". We had recent changes on bugfix that impact the XPT values, so the values you found are correct and the old ones are incorrect.

About the color, maybe the default color were changed, I don't know. Can you try changing it?

The configs are:
```
TFT_MARLINUI_COLOR
TFT_MARLINBG_COLOR

//buttons
TFT_DISABLED_COLOR
TFT_BTCANCEL_COLOR
TFT_BTARROWS_COLOR
TFT_BTOKMENU_COLOR
```

The default values are:
```cpp
#ifndef TFT_MARLINUI_COLOR
  #define TFT_MARLINUI_COLOR COLOR_WHITE
#endif
#ifndef TFT_MARLINBG_COLOR
  #define TFT_MARLINBG_COLOR COLOR_BLACK
#endif
#ifndef TFT_DISABLED_COLOR
  #define TFT_DISABLED_COLOR COLOR_DARK
#endif
#ifndef TFT_BTCANCEL_COLOR
  #define TFT_BTCANCEL_COLOR COLOR_RED
#endif
#ifndef TFT_BTARROWS_COLOR
  #define TFT_BTARROWS_COLOR COLOR_BLUE
#endif
#ifndef TFT_BTOKMENU_COLOR
  #define TFT_BTOKMENU_COLOR COLOR_RED
#endif
```

The available "named" colors (but you can use any hex value):
```cpp
#define COLOR_BLACK       0x0000  // #000000
#define COLOR_WHITE       0xFFFF  // #FFFFFF
#define COLOR_SILVER      0xC618  // #C0C0C0
#define COLOR_GREY        0x7BEF  // #808080
#define COLOR_DARKGREY    0x4208  // #404040
#define COLOR_DARKGREY2   0x39E7  // #303030
#define COLOR_DARK        0x0003  // Some dark color

#define COLOR_RED         0xF800  // #FF0000
#define COLOR_LIME        0x7E00  // #00FF00
#define COLOR_BLUE        0x001F  // #0000FF
#define COLOR_YELLOW      0xFFE0  // #FFFF00
#define COLOR_MAGENTA     0xF81F  // #FF00FF
#define COLOR_FUCHSIA     0xF81F  // #FF00FF
#define COLOR_CYAN        0x07FF  // #00FFFF
#define COLOR_AQUA        0x07FF  // #00FFFF

#define COLOR_MAROON      0x7800  // #800000
#define COLOR_GREEN       0x03E0  // #008000
#define COLOR_NAVY        0x000F  // #000080
#define COLOR_OLIVE       0x8400  // #808000
#define COLOR_PURPLE      0x8010  // #800080
#define COLOR_TEAL        0x0410  // #008080

#define COLOR_ORANGE      0xFC00  // #FF7F00

```

sjasonsmith, 2020-11-28:
> Those files cannot function on a stock Predator with the Trigorilla Pro board

The example we are referring to is for a Trigorilla Pro. What example are you using?

rq3, 2020-11-28:
> 
> 
> > Those files cannot function on a stock Predator with the Trigorilla Pro board
> 
> The example we are referring to is for a Trigorilla Pro. What example are you using?

The same example as you, under the bugfix configuration examples in delta/anycubic/predator. I'm not sure what machine it was configured for (I believe it was a Predator modified with an SKR board running Klipper, and then ported to Marlin), but it certainly won't run on a stock machine, and never has as published, for any version of Marlin.

thisiskeithb, 2020-11-28:
Here's the original PR by @TheNitek https://github.com/MarlinFirmware/Configurations/pull/157 which is based off https://github.com/SXHXC/Marlin-Anycubic-Predator-Trigorilla-PRO. There's some valuable information in the [readme](https://github.com/MarlinFirmware/Configurations/blob/import-2.0.x/config/examples/delta/Anycubic/Predator/README.md).

> I believe it was a Predator modified with an SKR board running Klipper

There's no mention of Klipper or an SKR board (the config uses `BOARD_TRIGORILLA_PRO`), so I'm not sure where that information comes from.

rq3, 2020-11-28:
> 
> 
> Here's the original PR by @TheNitek [MarlinFirmware/Configurations#157](https://github.com/MarlinFirmware/Configurations/pull/157) which is based off https://github.com/SXHXC/Marlin-Anycubic-Predator-Trigorilla-PRO. There's some valuable information in the [readme](https://github.com/MarlinFirmware/Configurations/blob/import-2.0.x/config/examples/delta/Anycubic/Predator/README.md).
> 
> > I believe it was a Predator modified with an SKR board running Klipper
> 
> There's no mention of Klipper or an SKR board (the config uses `BOARD_TRIGORILLA_PRO`), so I'm not sure where that information comes from.

Indeed, that was what got everyone excited about loading Marlin on the Predator. The Trigorilla Pro board is made by a Chinese company named Chitu, was closed source, and ran a really weird form of proprietary g-code.

Just knowing the physical process of jumper swapping to allow loading Marlin on the Trigorilla Pro got the ball rolling, BUT...
if you take a really close look at the video associated with that readme, you'll notice that it's in Russian (I think), and the file structure of the source computer references Klipper. At the end of the video, the display is upside down. Certainly no one ever ran that naked board in an actual Anycubic Predator. 

There are other references on the interwebs to that effort, which hint that the initial effort was to run Klipper, and that the Marlin configurations that resulted afterwards were an effort to "port" that over from an SKR board, AFTER they discovered how to load onto the Trigorilla.

Because I'm stupid, I took it all at face value, and eventually managed to get Marlin 2.0.5.3 to load and run on my stock Predator.
Between that revision and 2.0.7.2, there were too many changes to keep up with (I'm trying to print, not write code, which I truly suck at anyway), so I'm currently running 2.0.7.2, quite successfully.

Having said that, the example configuration files had (and have) errors which indicate that they could never have been used on a stock Predator. The temperature limits are dangerously high, the delta configuration has errors which won't allow calibration, the filament run out logic is reversed, the display is inverted, the touch screen didn't function, etc., etc., etc.

So. That's the history as I understand it from April of this year. I may be all wet, and probably am, but it really doesn't matter since I'm happily running Marlin 2.0.7.2.

I've even installed toggle switches on my machine so I don't have to do all that monkey motion with the board jumpers, and gleefully try every new nightly release of Marlin just for kicks. 

These days, I don't even have to physically rotate the display, and today I learned how to use the M995 g-code! Keep up the great work folks!

rhapsodyv, 2020-11-29:
@rq3 did you test the colours options?

Because I think we don't have a color issue, just a config changed.

rq3, 2020-11-30:
> 
> 
> @rq3 did you test the colours options?
> 
> Because I think we don't have a color issue, just a config changed.

Yes, I did, and I agree. Looking back through various Marlins, "u8g_dev_tft_320x240_upscale_from_128x64.cpp" has been modified several times (including its name). It's easy to mod the colors and buttons from there, but perhaps not apparent to the average user.

rhapsodyv, 2020-11-30:
Great. So I'm closing this, and I send a PR to Configurations repo, to fix the touch calibration values!

Thanks for your help!

closed 2020-11-30 by rhapsodyv