MarlinFirmware/Marlin #20313: [BUG] Building Problem : 'int freeMemory()' defined but not used [-Wunused-function]

Maker-M, 2020-11-28:
### Bug Description

Building problem when build the today bugfix 2.0 for the SKR mini E3 V2.0
The problem is : 'int freeMemory()' defined but not used [-Wunused-function]

### Steps to Reproduce

1. Download the last bugfix 2.0
2. Download the last bugfix or import configuration files (same problem for both)
4. Paste the config files from config\examples\Creality\Ender-3 Pro\BigTreeTech SKR Mini E3 2.0 to Marlin folder.
3. Run platformio
4. Change to default_envs = STM32F103RC_btt_512K in platformio.ini
5. Build

**Expected behavior:**

Fix problem

qwewer0, 2020-11-28:
I too get this warning message, many times within a single compiling:
```
Marlin\src\module\../inc/../HAL/./STM32F1/HAL.h:213:12: warning: 'int freeMemory()' defined but not used [-Wunused-function]
 static int freeMemory() {
```
SKR Mini E3 v1.2, `STM32F103RC_btt`/`STM32F103RC_btt_512K`
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5611004/Configuration.zip)

qwewer0, 2020-11-29:
@Maker-M You could test it, but I think this can be closed, as it fixed the issue on my end.

sjasonsmith, 2020-11-29:
Wait for it to be merged to close it.

Maker-M, 2020-11-29:
> @Maker-M You could test it, but I think this can be closed, as it fixed the issue on my end.

Ok I'll see that ! Many thanks

Maker-M, 2020-11-29:
> Wait for it to be merged to close it.

Yes I would like it to be merged too

closed 2020-11-29 by thinkyhead

Maker-M, 2020-11-29:
Thanks for the quick support

PR #20319 (2020-11-29): Fix unused freeMemory() warnings for STM32F1
### Description

STM32F1 builds have recently generated an `unused-function` warning from every file, due to the `freeMemory` function in HAL.h. This was most likley introduced due to changes to compiler arguments.

Declare this function as inline to prevent this warning, and remove a redundant cast while I'm touching the code.

### Benefits

Eliminates a massive number of compiler warnings.

### Configurations

Any STM32F1 will suffice. Here is the config I was using, although it has never been used in a printer.
[Configuration_adv.zip](https://git [... cut]