MarlinFirmware/Marlin #20317: Octoprint pausing 3d printer constantly with #define HOST_ACTION_COMMANDS enabled

RFBomb, 2020-11-29:
### Bug Description

My printer keeps pausing at the start of the print. When I get a print eventually started, it runs through the print without issue. At first I thought it was due to octoprint, but it has the same symptoms even when attempting to print from SD card. I have tested this in the follow scenarios:

- Printing freshly uploaded files on Octoprint
- Printing previously uploaded files on Octoprint
- Removing the SD card from the printer, manually dragging a file onto it via my pc, then attempting to trigger the print from SD via octoprint control. 

Printing with octoprint's USB cable unplugged (so no connection to the pi at all) prints fine. But if connected to octoprint, then this loop occurs about 30-40% of the time.

Note that this ONLY occurs when attempting to print. I can stream a file to the SD card from octoprint and it will upload the file without issue 100% of the time. 

When printing, the printer appears to enter a 'paused' state. Sometimes it will process some GCODE, other times it will not. 
I have uploaded a screencast to the AutoDesk site for viewing of the symptom here: https://autode.sk/2HM0BFs

My first thought was it is octoprint, but the issue occurred even when printing from SD card and Octoprint is simply attached. The issue began after enabling `#define ADVANCED_PAUSE_FEATURE `for M600 for filament runout, as well as the `#define HOST_ACTION_COMMANDS`.

My Octoprint does have a filament runout sensor installed, but its currently disabled. Octoprint enclosure is set up to trigger an M600, but this is not what the printer does typically when it pauses. It just sort-of sits there. Now, the symptom changes when I am printing from SD while connected to Octoprint. It will go into 'Nozzle Park' on the printer, but the printer just shows 'parking nozzle' on the screen, and the only way to resume it is using octoprint's 'resume' button. 

Sometimes (about 15-20% of the time) it will instead throw up the 'Purge more / continue' popup on octoprint enabled by `#define HOST_ACTION_COMMANDS`. When this occurs, there is a 50/50 shot of that prompt occuring on the printer, or just getirng stuck displaying 'parking nozzle' screen.

SO clearly there is something wrong with how this is interacting with octoprint / octoprint enclosure I think. 

@foosel (Octoprint) - @vitormhenrique (OctoPrint Enclosure) -- I'm tagging you guys since I have a feeling the issue pertains to some interaction going on here. 

### Configuration Files

Zip file contains a gcode sample, config files, octoprint logs / backup file (with any secret keys removed. But it can still be used to view the saved settings for the plugins).
[Printer Keeps Pausing.zip](https://github.com/MarlinFirmware/Marlin/files/5611662/Printer.Keeps.Pausing.zip)

### Steps to Reproduce

**Expected behavior:**

Print when i say print.

**Actual behavior:**

Print enters pause loop. See bug description section above for full details. 

#### Additional Information

- video of issue: https://autode.sk/2HM0BFs

- I had started another thread here: https://community.octoprint.org/t/various-issues-all-plugins-broke-octoprint-enters-a-pause-loop-before-print-before-even-telling-printer-to-start-heating-up/26070/3
The only response provided was 'probably corrupt sd card'. The main issue I started that thread about is resolved (all plugins seem to be working, and are interacting fine. Disk scans aren't reporting issues with the filesystem on sd card either. I will note that after I had done the first file system scan to see if any errors were found / could be fixed, that is when the plugins started displaying properly again, even though the report said no errors had been found. But the print pausing issue still persists.

vitormhenrique, 2020-11-29:
Have you tried removing all plugins from octoprint?
Just to make sure that it’s not a plug-in issue?

rhapsodyv, 2020-11-30:
Can you test bugfix?

And send the serial output, if the error continues? because the current logs says nothing about marlin, just octoprint thinking that you don't have filament on the printer, but no marlin data that allow us to see what could be happing.

thisiskeithb, 2020-11-30:
> Have you tried removing all plugins from octoprint?

You don't have to remove them. Just [start up Octoprint in safe mode](https://docs.octoprint.org/en/master/features/safemode.html) to disable plugins.

RFBomb, 2020-11-30:
The thread I linked to on the octoprint forum has that suggestion. I did
start in safe mode and it resolved the UI errors. A file system scan /
repair Resolved the UI errors permanently. While I was going that scan, I
also found a huge log file that left 0 space on the card. Deleted that
after the scan finished.

I just had to reinstall W10 due to a BSOD boot loop that started around 3
hours ago. So I have bigger fish to fry at the moment lol.

I’ll try another print once it’s up and running...

thisiskeithb, 2020-11-30:
Booting up OctoPrint in safe mode doesn't take long and is an easy thing to check to rule out plugins/language issues.

rhapsodyv, 2020-12-04:
@RFBomb did you test it again? Thanks!

RFBomb, 2020-12-05:
OK, so after a long (electronically challenged) week I've finally had a chance to test out printing again. 
I uploaded a file, and the pausing issue occurred. Octoprint told me update 1.5 was available, so after I cancelled the print, I disabled majority of my plugins, updated to 1.5, and restarted the pi. 

That file was able to print 4 times without having the pausing issue occur. So I will slowly begin adding plugins back in, starting with the ones I think are most important for my setup.

rhapsodyv, 2020-12-06:
I'm closing this, as it's a Octo print plugin issue. If you find anything else, you can reopen this. Thanks!

closed 2020-12-06 by rhapsodyv