MarlinFirmware/Marlin #20318: [BUG] Ender 5 Pro (MB 4.2.7) Not Working with BL Touch

BuddyBing, 2020-11-29:
### Bug Description

Unable to get BL Touch to work in any configuration (pin 27 board, Z or 5 pin locations).

### Configuration Files

[Configurations.zip](https://github.com/MarlinFirmware/Marlin/files/5611677/Configurations.zip)

### Steps to Reproduce

1. Recompiled Marlin using configuration example while enabling BL Touch (in many different configurations).

**Expected behavior:**

BL Touch interaction.

**Actual behavior:**

BL Touch does its double probing on bootup, but then becomes unresponsive after that.

#### Additional Information

Photos of how I have tried to have it plugged into the board.
[Photos (9).zip](https://github.com/MarlinFirmware/Marlin/files/5611679/Photos.9.zip)

ellensp, 2020-11-29:
you have #define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN when your probe is not plugged into the Z_MIN endsfop.
Disable this.
You should also use #define USE_PROBE_FOR_Z_HOMING if your using the probe as a z endstop.

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-11-29 by ellensp

BuddyBing, 2020-11-29:
Look at the other picture (as I posted a picture of two of the three configurations I tried).  I tried all three of the following configurations ( as well as the matching hardware configuration (ie. using the Z-End Stop, Using the 5 Pin, and also trying a configuration using a Pin 27 board and still had issues.

Here are the configuration: 
#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN

And tried this:

#define Z_MIN_PROBE_PIN 17 // Pin 32 is the RAMPS default

And tried this:

#define BLTOUCH
#if ENABLED(BLTOUCH)
  #define SERVO0_PIN27
   #define BLTOUCH_DELAY 200
#endif

tpruvot, 2020-11-29:
missing space ? #define SERVO0_PIN 27

ellensp, 2020-11-29:
Your bltouch servo plug is in backwards.
Blue wire should be on G pin.