MarlinFirmware/Marlin #20342: [FR] allow POWER_OFF_DELAY to function without AUTO_POWER_CONTROL + additional confirmation

bkiziuk, 2020-12-01:
### Description

https://github.com/MarlinFirmware/Marlin/pull/19987 brought the `POWER_OFF_DELAY` option.
This option depends on `AUTO_POWER_CONTROL`:

```
//#define AUTO_POWER_CONTROL      // Enable automatic control of the PS_ON pin
#if ENABLED(AUTO_POWER_CONTROL)
    //[...]
    //#define POWER_OFF_DELAY          60 // (s) Delay of poweroff after M81 command. Useful to let fans run for extra time.
#endif
```

However, even if `#define AUTO_POWER_CONTROL` is commented out, the M81 command functions and allows to cut the power to the printer.
So, the request is to allow `#define POWER_OFF_DELAY` to be used without the need of having `#define AUTO_POWER_CONTROL`

Additionally, it would be very convenient to have the option to cancel delayed auto-shutdown directly from printer's UI.

### Feature Workflow

1. for configuration: 
```
//#define AUTO_POWER_CONTROL
#define POWER_OFF_DELAY <seconds>
```

2. send M81 command
3. printer displays Cancel dialog with timeout
3. printer turns off after specified number of seconds if Cancel option was not executed

#### Additional Information

In my case, enabling `AUTO_POWER_CONTROL` causes immediate shutdown of the printer after it boots up.
Since main PSU also powers the logic board, turning off the power completely shut downs everything.

Board: MKS RobinNano 1.2
Power Control Board: MKS PWC 3.0

slowbro, 2021-07-07:
I'm not following what the issue is, perhaps it was fixed and the current code doesn't line up; what is triggering the M81 on startup? The only thing I can think of is that `AUTO_POWER_CONTROL` doesn't see a reason to keep the PSU on and the `POWER_TIMEOUT` is very low so it turns it off quickly..?

Can you verify this issue still exists with bugfix-2.0.x?

bkiziuk, 2021-07-07:
@slowbro let me look at it again. Lots of time have passed since it was opened.

bkiziuk, 2021-07-11:
@slowbro I think this issue can be safely closed. The feature seems to be working as expected. 
I also switched to control the entire process via end-Gcode, so there's no point in chasing it further.

closed 2021-07-11 by bkiziuk