MarlinFirmware/Marlin #20343: [BUG] UBL - G29 P1 stops at point 57

gbkwiatt, 2020-12-01:
I've seen same ticket, but closed and not resolved. 

I have SKR MINI E3 V2 board with bed size of 290x250

When I run on default UBL settings, printer HALTs on 57th point, and Mesh is invalid even after G29 P3.

 Current settings: `PROBING_MARGIN 10` and `MESH_INSET 1`.

I tried `PROBING_MARGIN 30` and `MESH_INSET 50` as as I have seen it as a solution. and it works fine.
I rather understand how to set it up to my bed.
How to calculate or set that so it probes as much bed as possible, I am struggling to understand it. 

Also at the end of G29 P1, nozzle goes to Z=0 in the middle of the bed which looks like it's touching the bed, pretty scary, is there a way to make it not to go to that position at the end of levelling?

boelle, 2020-12-01:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

gbkwiatt, 2020-12-01:
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5624687/Marlin.zip)
Here is my configuration and I am using latest marlin bugfix-2.0.x branch

ellensp, 2020-12-01:
Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information so that we're not just taking stabs in the dark. Here is the boilerplate:

- Download Marlin `bugfix-2.0.x` to test with the latest code.
- Enable `DEBUG_LEVELING_FEATURE` and `M114_DETAIL` and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun or Repetier Host.
- Send `M502` and `M500` to ensure your Configurations are applied.
- Issue the command `M111 S247` to enable maximum logging.
- Perform a `G28` to do your standard homing procedure.
- Do a `G29` to probe the bed. This will also enable bed leveling.
- Do some of the moves that revealed problems before. Take notes.
- Copy the log output into a **.TXT file** and attach it to your next reply.

Repeat this procedure, if needed, to demonstrate inconsistencies. From these logs we should hopefully get a better idea of what's going on with your machine.

gbkwiatt, 2020-12-01:
I am not having an issue, just asking for advise on how to calculate parameters to set for UBL to probe as much bed as possible

closed 2020-12-01 by ellensp